Ultimaker/Cura #4331: Cura x36 plugin not working.

LukeBonathan, 2018-09-06:
I'm trying to generate X3G file using the plugin available. Every-time I do a file generates to the desktop but its corrupt and get the error message below.

I've trued changing the printer in the cura profiles and a re-install.

**3.41**

**Mac - latest os**

**Flashforge**

**generate x3g file**

**generic error message**

**x3g file to be produced**

**Additional Information
[image]
**

nallath, 2018-09-06:
Could you also post the logs?

LukeBonathan, 2018-09-06:
Sure, here they are from my fresh install.
[cura.log](https://github.com/Ultimaker/Cura/files/2357620/cura.log)

nallath, 2018-09-06:
It's giving a permission denied error, which means that your operating system isn't allowing the plugin to open / use the executable it needs to actually do the conversion. What you need to do is change the permissions for the file located at 
"/Users/luke/Library/Application Support/cura/3.4/plugins/X3GWriter/X3GWriter/gpx_macos"

LukeBonathan, 2018-09-06:
Thank you for the tip, unfortunately changing the permissions doesn't seem to work still. I applied it to the whole cura folder and still get the error.

[image]

Ghostkeeper, 2018-09-08:
Next time, I suggest you report issues with this plug-in here: https://github.com/Ghostkeeper/X3GWriter/issues

After changing the permissions, did you get a different error in the log? Perhaps it's getting farther but still crashing somewhere else.

From what I can read, you can't add execute permissions from the interface in MacOS. Only with the terminal:

    sudo chmod +x gpx_macos

That's worth trying.

LukeBonathan, 2018-09-08:
Thanks, I will do in the future. 

Here is the latest log after trying the sudo command.

[cura.log](https://github.com/Ultimaker/Cura/files/2363104/cura.log)

diegopradogesto, 2018-09-11:
I also see this error in the log:
`2018-09-07 09:22:57,465 - ERROR - [MainThread] UM.Logger.logException [85]: FileNotFoundError: [Errno 2] No such file or directory: '/Users/luke/Library/Application Support/cura/3.4/plugins/X3GWriter/plugin.json'`

How did you install the package?

LukeBonathan, 2018-09-11:
Thats interesting. It was installed through the cura tool box. I took a look in the folder and found the directory doubled up. See pic.

I have moved the files back removing one of the x3g folders. I now get a new error in cura...
[image]
[image]

LukeBonathan, 2018-09-11:
[cura.log](https://github.com/Ultimaker/Cura/files/2372731/cura.log)
Here is the latest log with that error

TheFirstAxel, 2018-10-09:
I am still getting this error as well. Is there a fix?

miotislucifugis, 2018-10-09:
Same here... on several Macs running different versions of OS(sierra and High Sierra) and Cura (>=3.2)

diegopradogesto, 2018-10-16:
That is super weird. I only see these plugins being loaded:
```
2018-09-06 19:37:37,821 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [431]: Loaded plugin FileLogger
2018-09-06 19:37:37,832 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [431]: Loaded plugin LegacyProfileReader
2018-09-06 19:37:37,844 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [431]: Loaded plugin SimpleView
2018-09-06 19:37:37,856 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [431]: Loaded plugin SolidView
2018-09-06 19:37:37,869 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [431]: Loaded plugin FirmwareUpdateChecker
2018-09-06 19:37:37,881 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [431]: Loaded plugin SupportEraser
2018-09-06 19:37:37,900 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [431]: Loaded plugin PrepareStage
```

and then the X3GWriter has the error that I posted before. @TheFirstAxel  and @miotislucifugis can you also post the log files?

TheFirstAxel, 2018-10-16:
Sorry I saw that there was a version in the toolbox and updated it before I saw this. It is working now

diegopradogesto, 2018-10-17:
So, @LukeBonathan what about you. Are you still getting the error? In that case, can you please send the entire configuration folder? (You can zip it and attach it here)

Ghostkeeper, 2018-10-17:
I've actually posted an update for the X3GWriter at the beginning of this week. I hope it fixes a couple of things if you were using a printer without a predefined configuration for GPX, and fix the permission errors that MacOS users were getting.

The error in LukeBonathan's log is caused by Cura still registering old packages with outdated API version numbers but then not registering these plug-ins. This has been fixed in version 3.6: It should no longer show the option to save to X3G if the X3G writer is an outdated version that is no longer compatible.

So from what we've heard here, everything seems to be solved now.

closed 2018-10-17 by Ghostkeeper