Ultimaker/Cura #4332: [3.4.1] Initial layer temperature not set correctly when printing multiple objects one-at-a-time

hooyah, 2018-09-06:
**Application Version**

Cura 3.4.1

**Platform**

Windows10x64

**Printer**
Ultimaker Original+

**Steps to Reproduce**
* import any model
* duplicate and arrange
* set __Printing Temperature Initial Layer__ to something other then default
* set __Print Sequence__ to _one at a time_
* slice and look at gcode

**Actual Results**

in the gcode:  init. temp. is set correctly for the first object but then never again
every subsequent object will print with the normal print temperature (and potentially pop of the plate)

**Expected results**

The initial layer temperature should be set with each subsequent Layer 0
(and then reset to default again, too)

**Additional Information**

alekseisasin, 2018-09-13:
Hi @hooyah , you are right the initial temperature for all models should be same
DEVS: CURA-5722

Vandrasc, 2018-09-24:
Fixed! This will be added after the 3.5 release. 

@hooyah, thank you for reporting!

closed 2018-09-24 by Vandrasc