Ultimaker/Cura #4333: no reset after abort

teknomancer2008, 2018-09-06:
**Application Version**
ultimaker cura 3.4.1

**Platform**
windows 10 build 1803

**Printer**
prusa i3 (ctc clone)

**Steps to Reproduce**
load a project, attempt to print, press cancel print button on screen.

**Actual Results**
the program stops the print, but all the screen stays greyed out, and nothing is accessible as it is locked out.

**Expected results**
i expect to be able to actually continue as if i had nbot attempted to print in the fiorst place

**Additional Information**
this has caused me to constantly keep re loading the program just because of a single error
[image]

nallath, 2018-09-06:
I'm a bit confused. What cancel print button do you mean? Your screenshot doesn't seem to show what you are referring to.

teknomancer2008, 2018-09-06:
I was referring to abort print. In this shot, I have actually already pressed it. The screen remains greyed out like this, stopping further access. Not sure why...

fieldOfView, 2018-09-07:
You can switch back to the Prepare screen using the button in the top left corner.

alekseisasin, 2018-09-12:
So far as it is intended behavior I close the issue.

closed 2018-09-12 by alekseisasin