Ultimaker/Cura #4334: Custom G-code for Each Material Profile/More settings in Material Profiles

NicholasTracy, 2018-09-07:
Cura Version currently used is: 3.4.1

Printer is:    Tevo Tarantula

Firmware:    Marlin 1.1.8 with volumetric extrusion.

This is not a bug or issue, this is just a feature request. I am moving to Cura from Slic3r dev, in Slic3r dev there is custom G-code for Start, Stop, Filament Change, Nozzle Changes, as well as having custom G-code for each material profile. This is especially useful for Marlin where each material will have a different K value for the linear advance and possibly other material specific tuning parameters such as print speed, acceleration, jerk, filament diameter etc. It would be nice if Cura had a section in the material profiles where we could place material specific G-code and maybe include modifiers for thing like shell, wall, infill, first layer speeds and acceleration and jerk settings if used. As of now I have to manually enter the K value in to the G-code before sending it to the printer and I have to set up a different printing quality profile for each material I use to set the speed settings correctly. This leads to having a huge list of quality profiles with different quality settings for each material. instead this should be as easy as selecting the material and having only a few quality profiles to choose from. This way the material profile would have the relevant material settings like print speed or more.

nallath, 2018-09-07:
I doubt we're going to implement this. We used to let the machine handle a number of material settings (Eg; this is how the UM2 did it). Our idea was that the slicer would generate a path and the machine, depending on the material, would be able to change jerk, acceleration, temperatures, etc.
The main issue that we found with that is that in quite a few cases, depending on those settings, you do actually need to change the generated paths. If your slicer does not have different quality settings for that specific combination (eg; a specific profile for normal-pla so not just a normal + pla profile), this is impossible to do.

NicholasTracy, 2018-09-07:
I understand that this was handled by the machine for certain machines like the UM2. Cura is the go to software for many 3D printers. Much of them having the Marlin firmware. Marlin does not handle material profiles. I think at the very least there should be a material specific g-code section added to the material profiles so that users with custom machines can add in material related g-code just after the start g-code. This would allow things like Lin advance settings and other commands if we choose to use them. Other slicers have these features.

Ghostkeeper, 2018-11-08:
Since we do need to support multi-extrusion printers, a starting g-code specific to materials won't work (correctly). We can't just put the start g-code for every material at the beginning of the g-code, because if you have a second material that's something else, then the K-value for linear advance would always end up on the last material for the duration of the entire print.

There is a solution that we currently support though: You can adjust the extruder start g-code in a profile. This start g-code gets executed every time when switching to that extruder. To do this, you'd need to install the Machine Settings plug-in from the toolbox (by fieldOfView). Then you can edit the extruder start g-code in a custom profile.

The interface doesn't support editing this in the material profile. You can do that manually though by going to Help -> Show Configuration Folder... and under the `materials` subfolder, find your material profile and open it up with a text editor (or XML editor). Then next to the existing `<setting>` tags you can add a `<cura:setting>` tag that adjusts the `machine_extruder_start_code` setting to your preferred value.

I think the use case for adjusting the extruder start g-code in a custom material is really small, so I wouldn't be a proponent of implementing that (unless we decide on implementing adjusting all settings).

fieldOfView, 2018-11-08:
Note that there's also the Linear Advance plugin, which creates a linear advance setting in the sidebar. The value of this setting can then be used in the extruder start gcode snippet.

DeepwaterCreations, 2019-02-13:
I came here looking for this feature because I have some materials that print at temperatures colder than my Prusa Mk3's default minimum extrusion temperature. If I try to print them in their normal temperature range, nothing comes out of the nozzle. If I try to print them at 190, they string horribly or warp enough to make the print fail.

So far, the only way I know of to deal with this is to add gcode commands to adjust the minimum extruder temperature before the print happens. When I used Slic3r, I had this set up in the material settings for these materials. Now that I'm using Cura, I'm missing the feature terribly - I have to remember to edit my generated gcode by hand every time, and it's a pain in the butt.

Just wanted to throw that in here as another use case.

NicholasTracy, 2019-02-14:
My use case does not involve a multi extruder printer. But switching between materials would make this a nice feature to have.
Example: Switching between PLA and PETG which both have their own requirements for linear advance and retraction. As of right now Switching materials isn't enough, you have to create a seperate set of custom print settings for each material. So you have to change the material profile and the print quality profile. This is cumbersome. I really shouldn't have to explain why this should be a feature though. Just take a look at the way Slic3r does it.

NicholasTracy, 2019-02-14:
Accidentally hit the wrong button.

fieldOfView, 2020-03-23:
The Material Settings plugin (Marketplace: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MaterialSettingsPlugin) will let you set most settings per material. Though it won't let you edit the start- and end-gcode, it will let you edit Linear Advance and Retraction settings for each material (among many others).

Firedogs2x, 2021-12-27:
Hello, I know this is an older discussion, but I have a Modix printer that runs RepRap. I am learning about the "Pressure Advance" setting. So, I have been researching how to add the required M572 gcode. Since this setting can vary from material to material, I found myself here.
I am confused when GhostKeeper says "Then next to the existing `<setting>` tags you can add a `<cura:setting>` tag that adjusts the `machine_extruder_start_code` setting to your preferred value."
What does he mean next to?
This is what my code looks like at the settings tag.
```
<settings>
    <setting key="build volume temperature">28</setting>
    <setting key="surface energy">100</setting>
    <setting key="adhesion tendency">0</setting>
    <setting key="print temperature">232.0</setting>
    <setting key="heated bed temperature">80.0</setting>
    <machine>
```
I see the `<settings>` tag. Do I place the `<cura:setting>` tag directly above, below, or somewhere else? 
Or does it actually mean `<settings><cura:setting>`? 
Note: I know that the "M572 D0 S0.025" line needs to look like this. What I don't know is where do I place that line, and does it need all of the <setting key= format (as seen above)? 
Thank you very much for all of your help.

Ghostkeeper, 2022-01-03:
> What does he mean next to?

Indeed, you'll place them next to the other tags, in the same listing. The order doesn't matter, and they can be mixed. So you'd need something like:

```
<settings>
    <setting key="build volume temperature">28</setting>
    <setting key="surface energy">100</setting>
    <setting key="adhesion tendency">0</setting>
    <setting key="print temperature">232.0</setting>
    <setting key="heated bed temperature">80.0</setting>
    <cura:setting key="material_linear_advance_factor">2.0</cura:setting>
    <machine>
```

Note that the closing tag must also contain the namespace.

Perhaps an easier way to do this would be from the Cura interface when using the [Material Settings plug-in](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MaterialSettingsPlugin) by fieldOfView. I think it'd list the Linear Advance settings in the material manager then, too.

closed 2023-07-13 by github-actions[bot]