Ultimaker/Cura #4335: Default Print Temp not being displayed when ticked in custom view.

NicholasTracy, 2018-09-07:
**Application Version**
3.4.1

**Platform**
Windows 10

**Printer**
Tevo Tarantula (Marlin (volumetric))

**Steps to Reproduce**
- Created new custom printer.
- Set to single extruder.
- Settings visibility set to all.
- Create new material profile, set default print temp to 255.
- Select new material profile for extruder 1. 
- Go to manage printers, Extruder 1 tab in machine settings.
- Click on prepare.

**Actual Results**

Default material print temp is reading as 220 in machine settings > extruder 1 >materials
G-code generation also sets the default_material_temperature_layer_0 variable in start g-code to 220.
the default material print temp option is not visible in the material options menu even though the custom visibility settings has all of the check-boxes for material print temp checked. Can't change default print temp from 220 to 255, the material profile has default material print temp set at 255 but this has no affect and is being overridden by the now invisible quality profile setting.
**Expected results**
When a new material profile is created, the default temp set in the menu options should over ride the settings in the custom print quality profile unless the user has expressly changed this setting.
Also the default print temp, default print temp first layer, etc. options should be visible whenever they check-boxes are checked in the custom visibility settings.

**Additional Information**
[image]
[image]
[image]
[image]

NicholasTracy, 2018-09-07:
I seemed to have found my own solution. Although not convenient in any way. I removed the custom profile and basically created a new one based off of the Draft Quality profile which had the Default Printing Temp options visible. Now everything is fine. Although I have no idea why I experienced this issue in the first place and lost valuable time going through all my custom settings for my printer again.

NicholasTracy, 2018-09-11:
UPDATE The new profile that I created that seemed to fix the issue did the same thing, the Default Print Temp settings are all gone from the Materials configuration menu and Changing the temp in the material profile doesn't change anything when slicing.

alekseisasin, 2018-09-19:
Hi @NicholasTracy , I do not want to look unprofessional, reading the steps to reproduce I could not get your result. I was confused about this fact that settings were invisible. I had all of them. Just recently we upgraded Cura version to 3.5 Beta. Can you reproduce the same issue in Cura 3.5 Beta. If yes, then could you please make Numbered List reproduce them.

Ghostkeeper, 2018-11-16:
Maybe the g-code flavour was being set to Ultimaker 2?

NicholasTracy, 2018-11-29:
G-code flavor was set to Marlin(Volumetric E) This is still happening with the newest version of Cura.

1) All settings are checked as visible.
2) Printer is custom Marlin(volumetric E) printer with single extruder.
3) Material selected is a custom PLA from eSun Material temp is set at 220 in material profile.
4) Custom quality profile is created by duplicating one of the default ones while the custom material is selected. 
5) Create a quality profile from the first custom quality profile (changed a few settings like layer height and inner wall thickness but not temp)
6) After doing the above, the material extrusion temp settings are no longer visible in the settings of the custom profiles and the temp in the gcode is stuck at whatever the material extrusion temp was previously, if material profile temps are changed after this happens the gcode temps do not get updated to the new material profile settings.

pik33, 2023-03-10:
I cannot set the print temperature for Cura 5.3.0/Ultimaker2+
I cannot set the G-code flavor to Marlin, it automatically returns to Ultimaker when the printer is Ultimaker
I defined a custom material and set its print temperature to 225C, which is optimal for this kind of PLA, yet it didnt helped at all, the printer started printing at 195C which seems to be its default for PLA and 0.25 nozzle, but it is not good for THIS PLA. 

Please, do something and ENABLE THE PRINT TEMPERATURE SETTING for Ultimaker printers. The only workaround I have now is to wait for the printer to start printing and then manually set the proper temperature.

Multijeep24, 2023-11-04:
@NicholasTracy Have you found a solution? I've run through the same problem and can't find a solution.