Ultimaker/Cura #4341: Print one at a time mode won't slice items where gantry area overlaps bed area for 3rd party printers (but will on Ultimaker)

Chompworks, 2018-09-08:
**Application Version**
3.4.1

**Platform**
Mac OS X

**Printer**
Non-Ultimaker machines with gantry profiles set
[Tested on Ultimaker 2+ profile, and Could Not Reproduce]

**Steps to Reproduce**
Import any STL with a low-z height. e.g. calibration cube
Choose a non-Ultimaker printer profile, with gantry settings in place.
Move item(s) to edge of the bed so that the grey gantry safe visualiser overlaps the edge of the bed. If using multiple items, ensure their grey gantry safe zones are only overlapping the edge of bed and not each other. 

**Actual Results**
Item will not slice as item is considered out of build area

**Expected results**
Item should slice: Item will slice in exact same position if "All at once" mode is enabled, showing that the intended machine is capable of printing that part in that location. [See screenshots]
If profile is switched to an Ultimaker machine with gantry zone overlapping edge of bed (tested on UM2+ profile) item will also successfully slice.

**Additional Information**
[image]
[image]

alekseisasin, 2018-09-12:
Hi @Chompworks , if you cannot reproduce on Ultimaker 2 printer "[Tested on Ultimaker 2+ profile, and Could Not Reproduce]" then I assume it is not Cura issue. You can check your printer definition settings.
You can check the Settings: `machine_head_polygon` AND `machine_head_with_fans_polygon` 

**Be aware of changing settings of 3D Printer might damage your printer.**

Chompworks, 2018-09-12:
Funnily enough, could it be related to this?

https://github.com/Ultimaker/Cura/blob/master/cura/BuildVolume.py
Lines 437-439
>         # Find the largest disallowed area to exclude it from the maximum scale bounds.
>               # This is a very nasty hack. This pretty much only works for UM machines.
>              # This disallowed area_size needs a -lot- of rework at some point in the future: TODO

alekseisasin, 2018-09-12:
;), the commit was made in 2016, it is already a long time ago and we had a lot of changes, In my previous message I made the proposal for these settings because you changed _Print Sequence_ and we had similar for this case with 3rd-party printers.

Can you share your project, then I can check shortly?

Chompworks, 2018-09-12:
Sorry, was out at a clients

Here you go:

[CCR10_20mm Cube.3mf.zip](https://github.com/Ultimaker/Cura/files/2376513/CCR10_20mm.Cube.3mf.zip)
**Note - this has actually been Zipped, and is not just a 3mf file with the .zip suffix added.**

And the placement:
[image]

As you should see from the 3mf file
As soon as the grey gantry safe area overlaps the sides, it prevents the cubes from printing in One at a time mode. As you can also see, the cubes are in the printable area (even with reduced Z height) recognised by Cura, and they aren't overlapping one another.
In AIO mode, theres no such issue with them printing.
Similarly, when you move either cube away from the edges in OAAT mode, so that its central to the build plate, it will be sliced fine by cura.

That to me suggests theres a bug relating to code handling what happens when the gantry safe areas overlap edges. But only for 3rd party machines.

alekseisasin, 2018-09-14:
If you don't mind I will close the issue because the fix should be provided by 3d printer vendor, to do this they need to make a pull request with their changes.

closed 2018-09-14 by alekseisasin