Ultimaker/Cura #4342: tree support in thin air

mixtur, 2018-09-08:
**Application Version**
3.4.1

**Platform**
Windows 7

**Steps to Reproduce**
Use this model 
[doom_ed_v06.zip](https://github.com/Ultimaker/Cura/files/2363182/doom_ed_v06.zip)
 with Tree supports enabled

**Actual Results**
parts of the support are printed in thin air

**Expected results**
These icicles could be either cut with inclined wall, or have supports themselves. Former would take less filament, latest would probably be easier to implement.

**Additional Information**
[image]

mixtur, 2018-09-09:
Another kind of fun artifacts (though maybe that is more general case of icicles):
[image]

Here is the project:
[CCR10_doom_ed_v07.curaproject.3mf.curaproject.3mf.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2364519/CCR10_doom_ed_v07.curaproject.3mf.curaproject.3mf.curaproject.3mf.zip)

alekseisasin, 2018-09-12:
We will take a look. I can't promise that we will find fast solution for this issue.
DEVS: CURA-5717

mixtur, 2018-09-22:
One more case
[image]

And project:
[CCR10_doom_ed_v14.curaproject.zip](https://github.com/Ultimaker/Cura/files/2407816/CCR10_doom_ed_v14.curaproject.zip)

Ghostkeeper, 2018-11-21:
The second bug you're reporting should be fixed in 3.6. It's a duplicate of https://github.com/Ultimaker/Cura/issues/3249 (or at least caused by the same issue).

kgyrtkirk, 2019-01-30:
I was also running into similar issues with Cura 3.6 and found that reducing the value of "tree support branch diameter angle" could also help reducing "print in thin air" cases.
Thank you for the hint to raise infill density; in my case that will help a lot too.

jsheradin, 2020-12-01:
Just going to bump and say that I'm still getting the artifact shown in the first post here with version 4.7.

mereseydotes, 2021-06-09:
So anyway, I'm on Cura 9.1, will this ever be fixed?

Ghostkeeper, 2021-06-17:
Hopefully yes, by the time we get to Cura 9.1. Currently we're still on version 4.9 and 4.10 so it's going to take a while for the developers to catch up with you then.

We don't have a time estimate or planning for this issue. It's somewhat inherent in how Tree Support works, so we'd need to invent a new strategy if we are going to try to prevent this issue.

mereseydotes, 2021-06-17:
I tried 9.1 unless there's different number schemes. It seems worse. I caught it just flat printing in air without trying to expand a support