Ultimaker/Cura #4344: (3.4.1) File saved as .gcode comes up as STL.

chuckb93644, 2018-09-08:
**Application Version**
 3.4.1

**Platform**
Windows 10
[M_iphone_4_final.zip](https://github.com/Ultimaker/Cura/files/2363604/M_iphone_4_final.zip)

**Printer**
makerbotreplicator

**Steps to Reproduce**
After slicing, save to file (comes up as a gcode).  **Actual Results**
File copy comes up as STL.  When file is saved as .3mf, file copy comes up as STL.

**Expected results**
File saved as .x3g .

**Additional Information**
When saved file is opened with CURA, it is opened as an x3g file.
[M_iphone_4_final (2).zip](https://github.com/Ultimaker/Cura/files/2363605/M_iphone_4_final.2.zip)

diegopradogesto, 2018-09-11:
Probably a silly question, but... did you double check if the "Save as type" field in the dialog is "gcode"?

If you still have the problem, please send us the configuration folder, so we can try to reproduce the issue with the exact environment.

chuckb93644, 2018-09-12:
> Links:

Not a silly question.  I did Save as "gcode".  The saved file comes up as
"x3g" , but the Type column comes up as a "3D Model File".  As a newbe, I
am still working on it, so please put this thread on hold for now.  It may
turn out to be operator error, ( I have had more than one to my credit.  

Thanks for your response.

Chuck

https://github.com/

diegopradogesto, 2018-09-13:
It happens sometimes that you hover the "Save as type" dropdown and use the mouse wheel at the same time, then the type changes without noticing. 

I will close the ticket, but if you keep encountering the same problem, you can reopen it.

closed 2018-09-13 by diegopradogesto