Ultimaker/Cura #4348: Feature request : Variable fan speed based on layer time

asdasd-123, 2018-09-10:
Firstly, apologies if I haven't submitted this correctly, I don't usually use git hub and was taken here when searching "cura feature request" on google.

I believe a feature to vary cooling fan speed based on layer time would be a great addition. 

Currently when printing ABS, I have to keep an eye on sub-10 second layers otherwise the part overheats and droops. I usually manually set the fan speed at set layer heights to 20% to stop it from happening. 

I know there's currently the option to lift the head, but that slows the whole print down massively when you shouldn't need to.

I don't know the best way to approach it. but something along the lines of this for example : 
Lower bound layer time     -    0s
Lower bound fan speed     -    30%

Upper bound layer time    -     10s
Upper bound fan speed     -    0%

Which would mean that any layers under 10s have the cooling fan turned up progressively higher the quicker they are. Any any over 10s would have 0 cooling fan.

If not, no worries, I'll make a script to do it for me 👍, just figured it might be useful for others.

DDDirk, 2018-09-11:
There is also the option in Cura to set a minimum layer time, under Cooling. (You may have to enable that first under Preferences > Settings.) By default that is 5 s, so you could increase that.
That also slows the print down, but in a less abrupt way.
But that is really only a good option if the fan is already at full speed
So yesl, it seems like a useful addition to me.

asdasd-123, 2018-09-12:
Yeah I messed around with that, but that has three unnecessary issues with it.

1) It slows down the print time when having a fan turn on for small layers would allow you print normal speed

2) It heats up the part even more if you just slow the speed down during that layer which just makes it worse without a part cooling fan on.

3) If you instead raise the hotend, since you don't have a part cooling fan on you end up with oozing/stringing similar to dual nozzle printers, you'd need a purge tower or something to stop it

asdasd-123, 2018-09-12:
> Yeah I messed around with that, but that has three unnecessary issues with it.
> 
> 1. It slows down the print time when having a fan turn on for small layers would allow you to print at normal speed
> 2. It heats up the part even more if you just slow the speed down during that layer which just makes it worse without a part cooling fan on.
> 3. If you instead raise the hotend, since you don't have a part cooling fan on you end up with oozing/stringing similar to dual nozzle printers, you'd need a purge tower or something to stop it

asdasd-123, 2018-09-12:
No! I miss-clicked and closed it, please re-open it :(

NicholasTracy, 2018-09-12:
I believe there are options in the cooling menu that allow you to change minimum print speed and the layer time at which it slows down the print as well as a separate minimum layer time for fan speed and what fan speed to run at when a layer is below that time. Raising the nozzle is also an option when those fail and has its own settings as well.

DDDirk, 2018-09-13:
Silly me, I missed that: Cooling > Regular/Maximum Fan Speed Treshold. If the layer time goes below that value, "the fan speed gradually increases towards the maximum fan speed". That maximum fan speed can be set above it and it will be reached at minimum layer time.
See here:
https://ultimaker.com/en/resources/20421-cooling

asdasd-123, 2018-09-13:
Thanks guys, Looks like it sort of does what I need, but I'll read through all that link when I get back from work. 

it seems like a rather "bodged" way of making it fit the current UI though. I believe a separate section with "cooling profiles" which you can customise with a picture like that link shows would be a much better solution long term. You could pick them similar to how you pick a material in the top right currently.

diegopradogesto, 2019-09-11:
Closing this issue because the settings in Cooling do that. Thanks for sharing and thanks all for the replies and the support to @asdasd-123

closed 2019-09-11 by diegopradogesto