Ultimaker/Cura #4352: Cannot select item auto-placed off of build plate

mqudsi, 2018-09-10:
**Application Version**
Cura 3.4.1

**Platform**
Windows 10 x64

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
Auto import items that do not fit on the build plate

**Actual Results**
Some of the imported items are located off the build plate and cannot be selected with the mouse.

**Expected results**
The items should be selectable.

**Additional Information**
Upon importing the Cura-exported project file, I actually cannot even see these items. But if you open them with another viewer (I used the MS Print 3D metro app on Windows 10) you can see the items.

[Broken select off build plate.curaproject.zip](https://github.com/Ultimaker/Cura/files/2368175/Broken.select.off.build.plate.curaproject.zip)

nallath, 2018-09-11:
I can select the objects that were placed outside of the buildplate. Either with just clicking on it or with select all. 
Could you provide some more info about your system?

alekseisasin, 2018-09-19:
@mqudsi , for me Cura works well and I can select and drag and drop all models. You can select "Top View" and it will be easier to select the model. If you have still the issue then we need more information. You can record a short video from your screen.
[image]

Ghostkeeper, 2018-11-09:
I also can't reproduce this. It might have been a fluke or already fixed.

closed 2018-11-09 by Ghostkeeper