Ultimaker/Cura #4354: Tall support are unstable and almost useless

Bricksnspace, 2018-09-11:
Got an stl from thingiverse, https://www.thingiverse.com/thing:2122792

It is a barrel extension for a Nerf toy gun.

Cura generates support for side "grooves" but that support are unstable during printing. They are attached only on a very small area on bottom and on top, but during printing the support "wobbles" when extruder put a layer on it. At some point, one of them detached and fell off. 

Anyway, the object was perfectly printed, but supports are useless.
 

**Application Version**
Cura 3.4.1 (Linux x64 appimage)

**Platform**
Linux Fedora 28 (x86_64)

**Printer**
Creality3D Ender 3

**Steps to Reproduce**
Load object https://www.thingiverse.com/thing:2122792.
Select generic PETG material and "Draft Quality" profile. Check "Generate support" and "Build Plate Adhesion"
Click on "Prepare" button. 

**Actual Results**
Switch to "Layer view" and select layer in the middle.
[image]

Look at the support. It is "glued" to model only on top/bottom (start-end) layers of the side grooves.

**Expected results**
IDK. IMHO, generated support needs some lateral "link" to the object to increase stability.

**Additional Information**
A closer look to the lower side
[image]

Bricksnspace, 2018-09-11:
I'm a little confused. 
This was generated with 1.6mm horizontal expansion.
[image]
It is the intended result? Or I mess something?

fieldOfView, 2018-09-11:
Yes, this is the intended result. There is virtually no chance of that support falling over.

Bricksnspace, 2018-09-11:
Ok, I understand now. 
Thanks for your time :+1:

closed 2018-09-11 by Bricksnspace

fieldOfView, 2018-09-11:
Have you noticed that you can also set a negative value for Support Horizontal Expansion? That could remove the unneeded support altogether here.

Ghostkeeper, 2018-11-09:
Conical support was made to fix this issue.