Ultimaker/Cura #4355: Dual extruder - nozzle switch - not going to standby temp

kirilledelman, 2018-09-11:
When printing a 2 color object, extruder switch doesn't set previous extruder temperature to standby temp value. Both of the extruders remain at printing temperatures, causing massive amount of ooze from the inactive extruder and ruining the print.

**Application Version**
3.4.1

**Platform**
Mac

**Printer**
Custom FDM with 2 extruders

**Steps to Reproduce**
Set standby temp to 130, and printing temp to 195. Print 2-extruder print.

**Actual Results**
When extruder 1 finishes, temp remains at printing temperature, while extruder 2 prints.

**Expected results**
Extruder 1 temp should drop to 130 while extruder 2 prints, and vice versa for extruder 2.

**Additional Information**
I think it used to work in 3.4.0, 
Files attached: [Dice.zip](https://github.com/Ultimaker/Cura/files/2371404/Dice.zip)

nallath, 2018-09-12:
Confirmed; I've been able to reproduce this with master as well.

kirilledelman, 2018-09-13:
@nallath Is this a new problem in 3.4.1? I could maybe downgrade for now. Makes Cura unusable for me, as heat creep on stationary filament eventually creates a clog.

Ghostkeeper, 2018-09-14:
I see the behaviour but can determine that this is expected behaviour.

Cura prevents the nozzle from cooling down to stand-by temperature if it would cool down for less than a certain amount of time, by default 50 seconds. In this case you have the numbers of your die which print in way less than 50 seconds, so it decides that it shouldn't cool down to stand-by temperature.

The reason for this mechanism is that if you change the nozzle temperature often and significantly, the PID regulator that regulates the temperature in the nozzle tends to get out of whack and your temperature regulation becomes hugely inaccurate, overshooting by 50 degrees or so. We've even seen it develop a bias towards lower temperatures with every switch so that it eventually tried to print with a nozzle at room temperature.

The setting in Cura that governs this behaviour is a machine setting, meaning that it's normally not visible. You can install this plug-in from FieldOfView though to make them visible and reduce the `Minimal Time Standby Temperature` setting: https://github.com/fieldOfView/MachineSettingsActionPlugin
That should make it go to stand-by temperature even if the extruder prints very shortly. Heed my warning though!

closed 2018-09-14 by Ghostkeeper

fieldOfView, 2018-09-14:
Note that fieldOfView starts with a lowercase "f" ;-)

amagr0, 2018-09-14:
@Ghostkeeper misled me