Ultimaker/Cura #4357: 3.4.1 Cura Transform Mirror button is not present

Alde1980, 2018-09-12:
3.4.1
Windows 10 pro 64 bits
Problem persist in any printer configured
The button for mirror is not present

I have spanish language configuration, cura downloaded from oficial site, 4k monitor
[image]

fieldOfView, 2018-09-12:
Go to Help -> Show configuration folder. 
Locate the file named `plugins.json` in the configuration folder.
Post the contents of that file here.
Delete the file.
Restart Cura.

Alde1980, 2018-09-12:
thanks a lot for some reason it was disabled:

plugins.json content:
{"disabled": ["", "MirrorTool"], "to_remove": [], "to_install": {}}

Alde1980, 2018-09-12:
Sorry I thought it was fixes, after restart the file reappear with same settings and mirror is disabled

fieldOfView, 2018-09-12:
First close Cura, then remove the file, then restart Cura.
If you remove the file while Cura is still running, it will recreate the file when you close Cura.

Alde1980, 2018-09-12:
I remove the file after close cura and it creates it again with same parameters.
I have tried to make it read only and delete the "mirror" in disable parameter just in case.I t gives me an error an curo doesnt start. 
If I delete the file it reapears with same line after I reestart cura

fieldOfView, 2018-09-12:
Please post a link to cura.log in the same folder (eg on pastebin.com)

Alde1980, 2018-09-12:
Solved: In cura config where this  line : 
Disable Plugins: Mirror tool

I deleted and the button just shows, I dont know why it was dissabled in cura.cfg

fieldOfView, 2018-09-12:
It was disabled in a previous version of Cura.

Vandrasc, 2018-09-14:
Glad to hear there is no issue anymore, we will close this one then.

closed 2018-09-14 by Vandrasc