Ultimaker/Cura #4360: Dual Extrusion - Skirt lines and bottom of the Prime Tower

amagr0, 2018-09-12:
**Application Version** - CURA 3.4.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Select a dual extruder printer; select Skirt (both for extruder1); select "Prime Tower" and also "Ooze Shield"; analyse the toolpath.

**Actual Results** - One line of the "Skirt" is printed by the second extruder, when on the second extruder "Skirt" settings is select to be made by the "Extruder 1". And the bottom layer of the "Prime Tower" is also made by the second extruder. Since the second extruder will only start to print at layer 27 is not necessary to prime the second extruder through the print of the bottom of the "Prime Tower". 

BTW, why there is 3 lines "Skirt", if the "Skirt Line Count" is 2?

[image]

**Expected results** - Print the entire "Skirt" and also the bottom of the "Prime Tower" with the first extruder.

Thanks

[skirtlines.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2375442/skirtlines.curaproject.3mf.zip)

Ghostkeeper, 2018-11-13:
Currently the skirt is always printed with all used extruders. This is expected. There is one "main" extruder, for which the skirt line count is held as well as the skirt minimum length (whichever is greater). For all the other extruders, only the skirt minimum length is held. This is why you're seeing 3 skirt lines.

The functions of the skirt are these:
1. Make sure that the material is properly at the tip of the nozzle at the beginning of the print.
2. Make sure the material is able to flow, not clogged, and properly warm, early on in the print to save time in case it fails.
3. Make sure that the bed is level for all extruders, the extruders have the same height, etc.

You could consider not printing a skirt for one of the extruders, in your case, if:
* One of the extruders is not used in the first layer.
* Prime tower is enabled.

Because the prime tower will also ensure that 1. the material is properly at the tip of the nozzle, so for that the skirt will not be necessary any more.

It will also ensure that 2. the material is able to flow, but way too late. By layer 27, you'll have spent a lot of time, electricity and material only to discover that there is a flow problem in your nozzle.

It will not ensure that 3. the bed is at the correct height for the second extruder. If the Z calibration of your nozzles is off, you will not discover it other than in the final print.

So I think that the secondary skin line is still desired.

Liger0, 2018-12-24:
> Currently the skirt is always printed with all used extruders. This is expected. There is one "main" extruder, for which the skirt line count is held as well as the skirt minimum length (whichever [...]

What is the purpose of chosing the skirt material then?
Isn't there a way to print the skirt only with the decided material?

digitalfrost, 2022-07-27:
**Stale ticket.**
Can this be closed, since this refers to Cura 3.4.1 and we are now on 5.0 and the label is **Won't Fix/Do** ?