Ultimaker/Cura #4367: can't save to gcode file after recent upgrade to 3.41

davidstoll, 2018-09-13:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

davidstoll, 2018-09-17:
That is odd, I filled out all of that info when I posted it.  I don't know why it doesn't show.  Sorry about that.

Windows 10 64bit

It just simply has a little window at the bottom where it says that it is unable to save to that location.  But I can save other formats.  It's the same window that tells you when something is saved successfully or that there is a new version available to download.

I uninstalled the new version and re-installed the old one.  It saves gcode again just fine.  So, I upgraded again and had the same problem, so it wasn't a fluke.  I am back to the old version.

Let me know which log file (location/name) and I'll install the new version again so I can generate the log.

I'm using Monoprice MP Select Mini.

I'm using all the default settings.  I'm not changing anything.  Plus I've uninstalled and reinstalled.

v3.3.1 upgrading to v3.4.1

alekseisasin, 2018-09-19:
@davidstoll , the log file is located depends on your OS in:

    %APPDATA%\cura\<Cura version>\cura.log (Windows), or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
    $USER/Library/Application Support/cura/<Cura version>/cura.log (OSX)
    $USER/.local/share/cura/<Cura version>/cura.log (Ubuntu/Linux)

You might try the Cura3.5 Beta version. We made changes related to saving gcode. Could you try?

diegopradogesto, 2018-10-02:
Ping @davidstoll . Is this still happening to you? In that case, please provide the log file.

davidstoll, 2018-10-04:
Sorry for the delay.  I only found what looked like 3.5.0 (not beta).  I installed that, but get an openGL 2.0 error when I try to run it.  odd

diegopradogesto, 2018-10-05:
Can you send the logs, please?

davidstoll, 2018-10-05:
As I mentioned in my first post....Let me know which log file (location/name).  It is not obvious to me where it is, I am sorry about that.

diegopradogesto, 2018-10-05:
@alekseisasin wrote it many days ago. You can find the log file in those places depending on your OS

> @davidstoll , the log file is located depends on your OS in:
>
> ```
> %APPDATA%\cura\<Cura version>\cura.log (Windows), or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log [...]

davidstoll, 2018-10-05:
Ok, sorry, I didn't understand that was a path.  I thought that was a log or something.  Sorry.

[cura 3.4.log](https://github.com/Ultimaker/Cura/files/2450849/cura.3.4.log)

nallath, 2018-10-08:
Are you trying to write to local file or to SD? I don't see anything specific in your logs that indicate that something is failing. It does seem that you did something with the post processing plugin. Could you write down the exact steps to reproduce this (or make a video?)

davidstoll, 2018-10-09:
Does that log work?

davidstoll, 2018-10-09:
I've tried to save both to the SD and also the local HDD.

Simply download a premade file (stl or whatever), open in cura and save to gcode.  Editing (or not editing) the file makes no difference.

LipuFei, 2018-10-09:
@davidstoll Did you slice the model before you saved to gcode? If you open a mode file without slicing, there is no GCode thus nothing to be saved. And in Cura 3.4.1, by default it doesn't do auto slicing any more, so you need to click "Prepare" to slice after you load a model.

davidstoll, 2018-10-09:
No.  But, regardless, if I do the exact same thing in another version, it saves just fine.

LipuFei, 2018-10-09:
Hi @davidstoll , the log file you uploaded is a bit large and has a lot of messages in there, so it's pretty hard to figure out what went wrong. Could you remove your current log file `cura.log` and then try the following?
 - start Cura and try to save as gcode
 - when it fails, you can close Cura and upload that log file

You can find the log file in this location: `C:\Users\<your-user-name>\AppData\Roaming\cura\3.4\cura.log`. Thanks.

davidstoll, 2018-10-09:
I'm now getting an error when I try to run Cura.  OpenGL2.0 error pops up and I have to click ok.  Nothing happens after that.

LipuFei, 2018-10-09:
@davidstoll That could be related to your graphics card driver. Could you try upgrading your graphics card driver? Intel/Nvidia/AMD, etc.

davidstoll, 2018-11-07:
any other suggestions?

diegopradogesto, 2018-11-08:
Please send me the complete `3.3` user folder that is located in `C:\Users\<your username>\AppData\Roaming\cura` (you will need to zip it first). I'll try to do the same update that you did from 3.3 to 3.4 and see what happened. But the chance to found the cause is low since it looks that is dependent of your computer.

BTW, send me also the `3.4` folder that you can find in the same location.

davidstoll, 2018-11-08:
The folders are attached.  Thank you!
[3.3.zip](https://github.com/Ultimaker/Cura/files/2563053/3.3.zip)
[3.4.zip](https://github.com/Ultimaker/Cura/files/2563054/3.4.zip)

diegopradogesto, 2018-11-19:
Sorry @davidstoll it looks like you copied the folders from `C:\Users\<your username>\AppData\Local\cura` instead of from `C:\Users\<your username>\AppData\Roaming\cura` (notice **Local** instead of **Roaming**). Please attach the right folders.

davidstoll, 2018-11-24:
[cura3.3.zip](https://github.com/Ultimaker/Cura/files/2612009/cura3.3.zip)
[cura3.4.zip](https://github.com/Ultimaker/Cura/files/2612012/cura3.4.zip)

Sorry about that....Should be good now.  Thanks again for looking at this.

closed 2020-06-20 by no-response[bot]