Ultimaker/Cura #4368: Enhancement Request; Layer Height Decimal Limit in UI

CadDude, 2018-09-14:
I would like to bring to the attention of developers that the current value for layer height is restricted to 3 digits past the decimal place in cumulative z-stage settings in the g-code file.  However, I need to bring to the developers attention that the field for custom printer settings allows for a 4th digit after the decimal point that remains as entered.

My request is to only allow the UI to accept 3 digits after the decimal point or issue an alert to the user if they enter a 4th digit after the decimal point.

I did confirm that the last digit is ignored and the apparent error is cumulative.  I have been set straight as to understand this is not an error or a bug.  Simply a UI oversight.  There are probably other fields that this Enhancement Request could/should/or would apply to.

**Application Version**
Ultimaker Cura 3.4.1

**Platform**
Any

**Printer**
Any

**Steps to Reproduce**
Enter a custom layer height of 0.1925 (or any 0.nnnn).
Prepare the model.  
Save g-code to file
Open file with favorite editor
Calculate values from one layer to the next at least 3 times (to rule out rounding).
Compare the average layer height with the value provided by the user (still reading 0.1925)

**Actual Results**
Each layer to layer z-value calculates to 0.192.  This becomes a perceived cumulative error.

**Expected results**
I say perceived above only in my own ignorance that layer heights are in whole microns.
I would prefer the UI issue an alert regarding layer height values with a 4 decimal places -or- the UI excepts and shows only what will be used in g-code generation.

**Additional Information**
I've done this test and I know it is little thing.  Yet, the UI is the face of all the black magic under the hood.  Sometime it is good to know what the brew is made up of -exactly- what we put into it.

Thank you for considering this enhancement request.

fieldOfView, 2018-09-14:
Like I pointed out on the forum, even after 1000 layers (a 192.5 mm high print), the difference is only 0.5mm. That means you will get two more layers than you might have expected.

Sure, this can be fixed. But there are so many things that affect printing and/or usability so much more than this.

CadDude, 2018-09-14:
This is an enhancement request.  It goes on a list and gets prioritized. 
I for one was surprised to find the error and I am certain I will not be the only one.
Consider this at an entered value of 0.0509 for layer thickness.  It starts to matter.
In my day, you could validate field entry in the field definition.

mdealer, 2018-09-23:
To be honest I do not understand why arbitrary limitations even exist in Cura. The other day I was entering values for my printer that worked fine with Slic3r, but, no, Cura insists that I can't have high jerk or high travel speed or whatever else. The limits should be made configurable per printer...

nallath, 2018-09-27:
The limits *are* configurable per printer. It's just that whoever made the profile for the machine didn't do it. Also note that cura hardly ever prevents very high values; only if they are impossible (eg 100 m/s or something). Even then these can be adjusted per machine. 

As for limiting the resolution to 1 micron, that has nothing to do with the board, but with memory on your PC.

fieldOfView, 2018-09-27:
@nallath, that’s not at all what this issue is about; it is about the precission of what you enter in Cura being higher than what is used in CuraEngine.

nallath, 2018-09-27:
I know. I was responding to the remark about "arbitrary limitations" existing within Cura.

CadDude, 2018-09-28:
> <snip>
> As for limiting the resolution to 1 micron, that has nothing to do with the board, but with memory on your PC.

I regularly work and nanometer resolution and even angstrum calculation in CAD on a regular basis on a laptop PC.  These systems allow setting precision for general calculations and efficiency, but there is no real limit.  To the system, it is just a "unit". 

Even Meshmixer works at sub-micron levels on the Z-axis.  And I run that on a gutless Win10 laptop.

nallath, 2018-09-28:
Making it dynamic is always harder (and thus, more expensive). Since there are no machines that are anywhere near 1 micron yet, we set it to that. So, theoretically, you are right. In practice, not so much.

fieldOfView, 2018-09-29:
> A 300mm tall print at 0.0509mm intended layer height is significant by introducing a cumulative error of 5.4mm. (6000 layer at 0.050mm and truncating 0.0009mm = 5.4mm loss in height)

That's not how it happens. Instead of the 6000 layers at 0.0509mm (for the total height of 305.4), you get 6108 layers at 0.0509 to reach that exact same height. The model does not get truncated.

You are really making much more out of this than it is, and your fabricated worst case scenario still does not convince me there's a problem. But because I am done trying to explain this to you, and the change is literally one number in one line, I have made a PR to make sure that all float fields in the sidebar are now limited to 3 decimals after the dot. See https://github.com/Ultimaker/Cura/pull/4445

CadDude, 2018-09-29:
I tested it, IT DOES TRUNCATE!

commit 53b481ff (2018-11-01): Reduce the maximum precision of floats in SettingTextField inputs

No settings in Cura require more than 3 digits of precision. Many settings are in mm, and CuraEngine does its calculations in while microns. Having more precision in the frontend than in the backend apparently thoroughly confuses some people. Fixes #4368