Ultimaker/Cura #4379: Estimated print time is low with retraction

chputney, 2018-09-17:
**Application Version**
3.4.1

**Platform**
Windows 7

**Printer**
Custom printer - project file attacthed

The estimated print time that Cura calculates is usually quite accurate, but I have noticed that if I have a model that uses a lot of retraction the estimated print time is on the low side, sometimes by several hours.  Here is an example model attached.  The Cura estimated print time is 5 hours 2 minutes, but the actual print time was 6 hours 5 minutes.  It seems to me that Cura is not calculating the time for retraction properly.

If this model is sliced with no retraction, the Cura estimated time is 4 hours 51 minutes.  So Cura says that using retraction takes 11 minutes.

Looking is the gcode file for this model, there are 3931 instances of retraction, (They all start with "G1 F1500 E") and the retraction distance is 6.5mm.
So the total retraction time is: (3931) * (6.5) / (1500) = 17 minutes.

This isn't enough to explain the 1 hour discrepancy, but there seems to be an error somewhere in the retraction time estimate.   

[CP Desk Sign.curaproject.3mf.curaproject.3mf.curaproject.zip](https://github.com/Ultimaker/Cura/files/2390240/CP.Desk.Sign.curaproject.3mf.curaproject.3mf.curaproject.zip)

[CP Desk Sign with retraction.zip](https://github.com/Ultimaker/Cura/files/2390246/CP.Desk.Sign.with.retraction.zip)
[CP Desk Sign without retraction.zip](https://github.com/Ultimaker/Cura/files/2390250/CP.Desk.Sign.without.retraction.zip)

smartavionics, 2018-09-18:
Cura calculates the print time using what it believes are the accelerations the printer will use. If your printer firmware is using lower accelerations than what Cura thinks it is using, it will take longer to print than the estimate. Without looking at the code, I can't say whether it takes extruder acceleration into account or not.

diegopradogesto, 2018-09-19:
In Cura I see that the total time for the retractions is 20 minutes, but you have 9 minutes less of travels. So the calculation looks more or less correct. If you are experiencing 1 hour more of printing time it must be due to the extruder accelerations that are different depending on your firmware.

diegopradogesto, 2018-10-02:
I will close this issue since I don't see an issue here. If you think I'm wrong, please don't hesitate to reopen it again and provide other cases in which you see a big difference. In that case I can bring it to our control board to see if it's important enough to spend time in fixing it.

closed 2018-10-02 by diegopradogesto