Ultimaker/Cura #4381: Add option to see camera crosshair

kgyrtkirk, 2018-09-18:
**Application Version**
3.4.1

I would like to see the camera's pivot point (the point around which it rotates) if possible (like in openscad for example).
I sometimes like to check smaller details of the model - how it have been sliced; so I move and zoom to it; but after some of this I kinda loose the control of the camera.
Right now I use the reset camera to start over when I can't guess anymore where the focus is - adding an option to show the crosshair would probably make navigation easier.

Ghostkeeper, 2018-09-18:
As a workaround, enabling the preference "Center camera on selection" moves the camera's pivot point to your selection. We'll look into your suggestion though.

Ghostkeeper, 2018-09-18:
We discussed this within the team and the consensus was that the various ways to reset the camera position are sufficient to get control back. Basically the reset camera button you were using. We don't want to bloat the software too much. Sorry!

closed 2018-09-18 by Ghostkeeper