Ultimaker/Cura #4387: unable to update firmware ultimaker original plus using ultimaker cura 3.4.1

Pascal1573, 2018-09-19:
unable to update/upload firmware to ultimaker Original plus.
(automatic and custom)

**Application Version**
3.4.1

**Platform**
windows 10 build 1803

**Printer**
ultimaker Original plus

**Steps to Reproduce**
just update firmware

**Actual Results**
firmware update failed due to an unknown error
connection with printer broken after this action.
have to restart cura to fix.

**Expected results**
an updated printer

**Additional Information**

[image]

log file:
[cura.log](https://github.com/Ultimaker/Cura/files/2395923/cura.log)

alekseisasin, 2018-09-19:
Hi, @Pascal1573 I am not sure what problem it can be.
Recently we released a new Cura version 3.5 Beta (**it is not stable**) and you can try it. If you will do it then probably you will need to save Cura settings to another folder, just in case if we will have an issue in the version upgrade. The Cura settings you can find here:
    %APPDATA%\cura\<Cura version>\cura.log (Windows), or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
    $USER/Library/Application Support/cura/<Cura version>/cura.log (OSX)
    $USER/.local/share/cura/<Cura version>/cura.log (Ubuntu/Linux)

You mentioned Cura 3.4.1. You might try Cura 3.3 version to upgrade firmware.
Where did you get your custom firmware? Did you do some custom changes in firmware or updated before?

Pascal1573, 2018-09-19:
Okay, I will try an older cura. (and the 3.5 beta.)
I had this problem once before, unfortionatly I don't remember whitch version of cura this was.
Will search the older thread and, as mentioned, will try older version.
I don't update with every cura release, but wanted to try something today .
That's how I found this one.

Pascal1573, 2018-09-19:
@alekseisasin .
see this thread:
https://github.com/Ultimaker/Cura/issues/2028

Pascal1573, 2018-09-19:
okay..

cura 3.5 beta.   not working firmware update
cura 3.3.1         not working firmware update
cura 3.1.0         firmware update works.

I had the firmware loaded from march 2018, that's the one included with cura 3.3.1.
So I think that it worked with that version in the past.
The only change between installing cura 3.1.0 and cura 3.3.1 is "THE"windows 10 64bit update to 1803.
So that can be one of the causes too.
……..????…….

GMagician, 2018-09-21:
@Pascal1573 that sounds too bad. I raised issue times ago, #3731, and it should have been fixed by #3999 but it seems it isn't...

fieldOfView, 2018-09-21:
#3999 fixed a different problem with firmware updating.

To get to the bottom of this, could someone try the following for me?
* download & install Cura 3.5
* download https://gist.github.com/fieldOfView/a62f1fbadd9f57ac9f16f4281d67e01b
* replace C:\Program Files\Ultimaker Cura 3.5\plugins\USBPrinting\USBPrinterOutputDevice.py with that file
* restart Cura and try a firmware upgrade (it will still fail)
* locate the cura.log file (from the Configuration folder) and post a link to it here.

Pascal1573, 2018-09-21:
@fieldOfView 
Going to try it now.

Pascal1573, 2018-09-21:
@fieldOfView 

And done. logfile added.
[cura.log](https://github.com/Ultimaker/Cura/files/2406228/cura.log)

GMagician, 2018-09-21:
> #3999 fixed a different problem with firmware updating.

@fieldOfView you are right

fieldOfView, 2018-09-22:
@Pascal1573 can you try another computer perhaps? Since it did seem to have worked with 3.3 at one time, but no longer, that make your computer/OS suspect. And if you reinstall Cura 3.1, does it work then?

There was no obvious change between 3.1 and 3.3/3.4 that would cause a timeout now, as far as I can see.

Pascal1573, 2018-09-22:
@fieldOfView  Hi Aldo now I am a bit confused. Earlier this week I have been able to load the firmware with v3.1.0. Now I have indeed been able to change the firmware, after a reinstallation of v3.3.1, to the 15th of March 2018. In other words the firmware can be upgraded again with v3.3.1.

I will do the same thing again with v3.4.1, but I'm pretty sure I installed this virsion after windows was updated to build 1803.

Please take my word for it, that I did do the same thing before posting this thread. That is the confusing part for me at this moment, if it work with v3.4.1 after the reinstall that I going to do now.

Pascal1573, 2018-09-22:
@fieldOfView   Hi Aldo, Normaly I understand a lot... Now I realy confused. It now works with v3.4.1

Sorry!! I don't understand why…. (yes perhaps the reinstallations ofcura ..) but why?
Firmware uploaded, unfortionatly the fw date is the same mach 15th 2018.. But it uploaded.

Again.. sorry for starting this thread!

fieldOfView, 2018-09-23:
No problem at all. You can close this issue for now, or some UM dev will do so, but be sure to reopen it if the problem resurfaces. Problems that magically go away have a tendency to resurface.

closed 2018-09-23 by Pascal1573