Ultimaker/Cura #4388: Separated Horizontal expansion for Outer and inside edges ( holes)

djvethaak, 2018-09-19:
For all **mechanical designer**s it's very annoying that we have to adapt our model to the slicer cura.
Currently there is only one horizontal expansion which changes the hole size but also the outside of the model. To get the correct size you need to use scale up and use the horizontale expantion for holes.

We as mechanical engineers we would love to see that the horizontal expansion will be divided in to 2 different sub settings. The horizontal expansion of the **outer side**  and an horizontal expansion for **holes**.

See drawing at the end.

Yes , this uses some math. But there is a way to check it quickly by checking the edges on  x y search raster on each z layer . **See drawing.** 
 
It takes some time , but engineers would love to wait a bit and get a correct print directly.

**Application Version**
Cura 3.4.1

**Platform**
um3

**Printer**
um3

**Steps to Reproduce**
make raster bigger ( setting) and when possible hole found , check at smaller raster . For z layers above , mostliky the same holes are there ... 

**Actual Results**

**Expected results**
_A model with perfect outside dimensions and correct hole size._ 

**Additional Information**
[image]

alekseisasin, 2018-09-19:
Hi @djvethaak , I don't want to disappoint you but probably we cannot implement the requested functionality. Cura is more CAM software application and does not provide extended functionalities for making changes in your model. I understand sometimes simple work is boring and takes more time than expected. 
As I remember the latest CAD applications allow to use variables for measurements and these variables can be changed simply in the table.  It helps to save time and reduce repetition in your working tasks.

djvethaak, 2018-09-19:
Hi , i don't want to change my model in cura , just want that the slicer implement those settings so that only and only the g code output file is modified.

DDDirk, 2018-09-19:
In your first example, there is no (fully enclosed) hole. But you still might want to fit something in there, it might be a slot. Only you as the designer would know that. You wouldn't want Cura to make a wrong guess at what you want.

alekseisasin, 2018-09-19:
It can be very tricky applying this setting and let Cura make decisions. I see a better solution when a user might apply settings for certain surfaces. Then it will prevent the wrong "changes" in the model.

DDDirk, 2018-09-20:
It is still an issue, though. It makes sense to want to design something the way it should turn out. However, depending on the production method, the model may need to be adapted. A bit like changing a text-file to PDF just before you print it, so you can better control the way it comes out. But you shouldn't do any original editing in PDF because that is hard to transform into HTML if you want to put it on a website. In other words, you need a general file and then when you're ready for 'production' adapt it to the production method.
The question is then, in this analogy, should Cura be like a PDF-editor? Or should that be a separate step before you send it to Cura? The former is more user friendly (in most cases), the second is more flexible.

Ghostkeeper, 2018-11-16:
I'd say that this falls under "preparing your model for printing" which is within the scope of Cura.

The main problem for us with implementing this sort of thing, is that performing offsets is quite a difficult algorithm to get right in every case. We currently delegate that to [ClipperLib](https://github.com/Ultimaker/CuraEngine/tree/master/libs/clipper) which devotes some ~1000 lines of code to the problem, and has had a good 7 years of development to work out the bugs in this implementation. But Clipper doesn't support offsetting outer angles with a different value than inner angles. So we'd have to implement that ourselves.

closed 2023-07-13 by github-actions[bot]