Ultimaker/Cura #4389: Improvement Request - Dual Extruder Toolchange Temperatures

amagr0, 2018-09-19:
**Application Version** - CURA 3.4.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Select a dual extruder printer; select "Prime Tower"; analyse the gcode.

**Actual Results** - On the toolchange action the extruder that was not printing is first to head up the temperature, and when it reached the desired temperature the other extruder will cool down. 
Since the extruder that starts cool down did not reach the "Standby" temperature when the active extruder starts to print, this will leave some oozing material on the printed part. 

**Expected results** - During the toolchange, first cool down the active extruder with a M104, and then head up the temperature other extruder with M109, this will avoid contamination by the inactive extruder on the printed part. Or tha last command on the active extruder it could be a M104 S("standby temp").

[image]

amagr0, 2018-09-19:
The code M104 S(material_standby_temperature) on the "Extruder End-Gcode" is not working, any ideas why? Tks

[image]

[image]

[image]

zerspaner-gerd, 2018-09-19:
Try **{material_standby_temperature,` 0}**
The number after the comma is for the extruder.
That works for me.

amagr0, 2018-09-20:
I use () rather that {}.
It's not necessary to declare the number of the extruder.
Tks

fieldOfView, 2018-09-20:
It is not working _because_ you use () instead of {}. You should be using {}, then the replacement works. Supplying the extruder number is indeed optional.

Ghostkeeper, 2018-09-21:
As for the original issue... This is actually intended!

Cura's heating and cooling speed predictions are not perfect. Indeed, they assume that the temperature increases or decreases linearly while they tend to heat up slower when they are hot and cool down slower when they are cold.

We found that when Cura's predictions are wrong, the predictions become instable. This is the reason: If Cura predicts that the nozzle heats up faster than it actually does, then the `M109` will pause for a while (if the predictions would be perfect, `M109` would be a no-op). During this time, with your proposed change, the first extruder would be cooling down towards 170 degrees due to the `M104 To S170` command. So if our prediction is further wrong, the other extruder gets the chance to cool down further. And when that nozzle is cooled down further, our new prediction for the next extruder switch will be even further off.

That's why we perform the M109 before the M104 now, to get a stable equilibrium instead of an unstable one. This is the only place where we can sync this up, since placing the M109 halfway through the extruder plan to sync up the stand-by temperature would cause the nozzle to pause while the other extruder is printing.

closed 2018-09-21 by Ghostkeeper