Ultimaker/Cura #4390: [master] cura crashes on startup with latest master branch

smartavionics, 2018-09-19:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

If I try and use the master branch that's more recent than a870060a34a61367f0, Cura crashes on startup  with messages like these:
````
2018-09-19 13:42:39,048 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [54]: Loading container Kossel XL_user
2018-09-19 13:42:39,051 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [374]: Container [Kossel XL_user] added.
2018-09-19 13:42:39,052 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [54]: Loading container Kossel XL_settings #9
2018-09-19 13:42:39,054 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [374]: Container [Kossel XL_settings #9] added.
2018-09-19 13:42:39,055 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [54]: Loading container kossel_xl
2018-09-19 13:42:39,091 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [59]: Exception in function (maximum recursion depth exceeded during compilation) for setting: 0.8 * min(extruderValues('machine_nozzle_size'))
2018-09-19 13:42:39,093 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [59]: Exception in function (maximum recursion depth exceeded during compilation) for setting: 0.8 * min(extruderValues('machine_nozzle_size'))
2018-09-19 13:42:39,094 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [59]: Exception in function (maximum recursion depth exceeded during compilation) for setting: 0.001
2018-09-19 13:42:39,095 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [59]: Exception in function (maximum recursion depth exceeded during compilation) for setting: 0.1 + 0.4 * machine_nozzle_size
2018-09-19 13:42:39,095 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [59]: Exception in function (maximum recursion depth exceeded during compilation) for setting: 2 * machine_nozzle_size
2018-09-19 13:42:39,096 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [59]: Exception in function (maximum recursion depth exceeded during compilation) for setting: line_width
2018-09-19 13:42:39,096 - ERROR - [MainThread] UM.Settings.SettingFunction.__init__ [59]: Exception in function (maximum recursion depth exceeded during compilation) f
````

I have tried removing the cache but it made no difference.

The Kosssel XL printer is defined like this:
````
{
    "id": "kossel_xl",
    "version": 2,
    "name": "Kossel XL",
    "inherits": "fdmprinter",
    "metadata": {
        "visible": true,
        "author": "Mark Burton",
        "manufacturer": "Various",
        "category": "Other",
        "file_formats": "text/x-gcode",
        "icon": "icon_ultimaker2",
        "platform": "",
        "platform_offset": [0, 0, 0],
        "machine_extruder_trains":
        {
            "0": "kossel_xl_extruder_0"
        }
    },
    "overrides": {
        "machine_heated_bed": {
            "default_value": true
        },
        "machine_width": {
            "default_value": 240
        },
        "machine_depth": {
            "default_value": 240
        },
        "machine_height": {
            "default_value": 300
        },
        "machine_center_is_zero": {
            "default_value": true
        },
        "machine_shape": {
            "default_value": "elliptic"
        },
        "machine_nozzle_size": {
            "default_value": 0.4
        },
        "material_diameter": {
            "default_value": 1.75
        },
        "machine_gcode_flavor": {
            "default_value": "RepRap (RepRap)"
        },
        "machine_start_gcode": {
            "default_value": "G28 ; home all axes\nG1 X0 Y0 Z10 F8000 ; descend\n; run print prologue\nM98 P/sys/printstart.g\n"
        },
        "machine_end_gcode": {
            "default_value": "M83 ; extruder relative mode\nG1 E-2 F2000 ; retract\nM104 S0 ; turn off extruder\nM140 S0 ; turn off bed\nM106 S0 ; turn off fan\nG28 ; home\nM84 ; disable motors\n"
        },
        "machine_max_feedrate_z": {
            "default_value": 300
        }
    }
}

````

And it references an extruder definition like this:

````
{
    "id": "kossel_xl_extruder_0",
    "version": 2,
    "name": "Extruder 1",
    "inherits": "fdmextruder",
    "metadata": {
        "machine": "kossel_xl",
        "position": "0"
    },

    "overrides": {
        "extruder_nr": { "default_value": 0 },
        "machine_nozzle_size": { "default_value": 0.4 },
        "material_diameter": { "default_value": 1.75 },
        "retraction_amount": { "default_value": 2 },
        "retraction_speed": { "default_value": 20 }
    }
}
````

smartavionics, 2018-09-19:
The master branch up to 4777c3f41df76cd is OK so my guess is that it is ddc5d983a1625c3 that is causing the problem. What do you think @LipuFei ?

LipuFei, 2018-09-19:
@smartavionics Let me check your definition files.

smartavionics, 2018-09-19:
OK, thanks, they could easily be bad, I know nothing about how they should look.

LipuFei, 2018-09-19:
@smartavionics If I start `master` clean with your definition files and then add Kossel XL, everything seems to be fine. Could you try removing your cache folder? On Linux, it's `$HOME/.cache/cura`

smartavionics, 2018-09-19:
That was the first thing I tried, didn't make any difference.

LipuFei, 2018-09-19:
Could you send me your configuration folders then? Thanks!
 - `$HOME/.local/share/cura`
 - `$HOME/.config/cura`

smartavionics, 2018-09-19:
I'm just giving you the master sub-directories as there's a lot of other older rubbish which you don't want.

[local-share-cura-master.zip](https://github.com/Ultimaker/Cura/files/2397355/local-share-cura-master.zip)

[cura-config-master.zip](https://github.com/Ultimaker/Cura/files/2397356/cura-config-master.zip)

Thanks

LipuFei, 2018-09-20:
@smartavionics Sorry that I still cannot reproduce this problem... For a clean start, what I did was:
 - Create an extruder definition file named as `kossel_xl_extruder_0.def.json` in `resources/extruders`. The content is what you posted above.
 - Create a machine definition file named as `kossel_xl.def.json` in `resources/definitions`. The content is what you posted above.
 - Remove all the following folders:
    - `$HOME/.cache/cura`
    - `$HOME/.config/cura`
    - `$HOME/.local/share/cura`
 - Start Cura master and add Kossel XL.

smartavionics, 2018-09-20:
Good Morning. Understood, but what I am doing is trying to load cura with my existing printers + profiles in place. I'm not creating a fresh instance of Kossel XL.

LipuFei, 2018-09-20:
Ah, I see that you have some machine and extruder definition files in your `.local/share`:
 - `definitions/kossel_mini_mb.def.json`
 - `definitions/kossel_xl.def.json`
 - `extruders/kossel_mini_mb_extruder_0.def.json` (*)
 - `extruders/kossel_xl_extruder_0.def.json` (*)

Those files could cause some problems in the version upgrade, especially the two extruder definitions. You could try moving them into the Cura source code directories in `resources/definitions` and `resources/extruders`

smartavionics, 2018-09-20:
I would prefer to keep them in .local/share if possible so that they are independent of the Cura installation in use.

LipuFei, 2018-09-20:
I managed to get some weird errors. I'll keep digging.

LipuFei, 2018-09-20:
@smartavionics Found one thing with your Kossel Mini definition. You Kossel Mini extruder definition is named as `kossel_mini_mb_extruder_0`, but in `kossel_mini_mb.def.json` it has this:
```
"machine_extruder_trains":
{
    "0": "kossel_xl_extruder_0"
}
```

This needs to be changed.

smartavionics, 2018-09-20:
OK, thanks.

LipuFei, 2018-09-20:
Let me know if it works. I'm getting some segfaults now but that's a different problem.

LipuFei, 2018-09-20:
@smartavionics I've pushed 2 fixed into Uranium `master`. Tested with your local directories + the fix for kossel mini, looks good here.

smartavionics, 2018-09-20:
I made the change in Kossel Mini and Cura started OK so I guess that was the main issue. Sorry, that was my fault being dozy! If you're happy, lets close this issue now.

closed 2018-09-20 by LipuFei