Ultimaker/Cura #4392: Z-hop not working over empty bed.

YerBabiE, 2018-09-20:
**Application Version**
Latest v3.4.1

**Platform**
Mac OS

**Printer**
anet A8

**Steps to Reproduce**
z hop feature does not work over blank bed, so nozzle if putting streaks on bed which is transfering to the 1st layer.

**Actual Results**
my mirror finish is not good on first layer.

**Expected results**
Z hop should treat the bed like the rest of the print if there will be filli going down.

**Additional Information**

smartavionics, 2018-09-20:
You don't provide any useful information like a cura project or screen shots or gcode so it's difficult to understand exactly what is happening but if you try the new cura 3.5 beta and use the no skin combing mode you may find that you get better results.

YerBabiE, 2018-09-20:
yer, I don't know what to say, the gcode / project does not matter, the problem is pretty basic, z hop does not work over a "blank" bed, only printed areas, so if that area of the bed is going to be printed on (first layer) then it can be market by a move prior to this first layer going down.

smartavionics, 2018-09-20:
OK, but if you don't provide any useful info, you can't expect to get your issue resolved. If you provided me with a project file (.3mf) for some example I could load that here and look into what is happening and whether more recent cura releases have already fixed your issue.

Ghostkeeper, 2018-11-16:
Disable the setting: "Z Hop Only Over Printed Parts".

closed 2018-11-16 by Ghostkeeper