Ultimaker/Cura #4394: [3.5]Importing trimesh does not work on frozen builds

fieldOfView, 2018-09-20:
**Application Version**
3.5 beta

**Platform**
Windows, likely others as well

**Printer**
n/a

**Steps to Reproduce**
* create a plugin that imports trimesh
```import trimesh```

**Actual Results**
The plugin won't load (with an ImportError)

The frozen build has the following unmet dependencies:
* decorator (https://pypi.org/project/decorator/)
* packaging (https://pypi.org/project/packaging/)
* pyparsing (https://pypi.org/project/pyparsing/)

Some required resources are also stripped from the frozen build, namely trimesh/resources/*.template

There's also a soft dependecy on rtree (https://pypi.org/project/rtree/) for some functionality in trimesh that would be very nice to have, but this is not blocking.

**Expected results**
Importing trimesh works and the plugin loads

Appesteijn, 2018-09-21:
CURA-5752

Appesteijn, 2018-09-21:
This also blocks your MeshTools plugin

LipuFei, 2018-10-02:
@fieldOfView Do you have a plugin that I can test with? I'm creating new 3.5 builds now with trimesh dependencies.

Appesteijn, 2018-10-02:
It is in the contribution portal, I'll send you a link

fieldOfView, 2018-10-02:
For reference: https://github.com/fieldOfView/MeshTools

closed 2018-10-03 by diegopradogesto