Ultimaker/Cura #4395: [3.5 Beta] Input filament diameter?

zerspaner-gerd, 2018-09-20:
**Application Version**
3.5 BETA

**Platform**
Windows 10 64bit

**Printer**
UM2+ with marlin 
[test.zip](https://github.com/Ultimaker/Cura/files/2401536/test.zip)

**Steps to Reproduce**
No matter with all printers the same.

I can not find a way to specify the filament diameter!
The attitude is gone!
[image]

Am I the only one with this mistake?

fieldOfView, 2018-09-20:
It has gone to the same category as the Nozzle Diameter, so it is no longer available in the sidebar, unless you install the Printer Settings plugin from the Toolbox.

zerspaner-gerd, 2018-09-20:
I do not understand

Where should I now enter my measured filament diameter?

fieldOfView, 2018-09-20:
On the Materials pane of the preferences. Or, if you want to have a setting in the sidebar, you need to install the Printer Settings plugin from the Toolbox, restart Cura, and look for the material diameter setting under the new "Printer Settings" category (which also has a nozzle size setting, wohoo!)

SteveDC2, 2018-10-15:
I have an issue with this plugin...
I saved a project before installing the plugin and if I now open that project the "printer settings" category disappears.

Is this a bug or is there something I need to do to have "printer settings' appear in the settings when loading a project?

fieldOfView, 2018-10-15:
If you have another issue than the opening post, please open another issue.
The PrinterSettingsPlugin has its own issue queue here:
https://github.com/fieldOfView/PrinterSettingsPlugin/issues

SteveDC2, 2018-10-17:
OK, will do.