Ultimaker/Cura #4396: Adaptive layering interferes with support structure generation

kulfuerst, 2018-09-20:
**Application Version**
3.4.1

**Platform**
Windows 10

**Printer**
Anycubic i3 Mega

**Steps to Reproduce**
Enable adaptive layers, enable support generation, set support overhang angle to 60% or lower

**Actual Results**
size of generated support structure is greatly reduced when adaptive layering is enabled, in spite of a set support overhang angle.

**Expected results**
No change in generated support structures

**Additional Information**
[image]
[image]

nallath, 2018-09-20:
Unfortunately, I don't think there is much we can do to fix this. The reason this happens is because of the way we actually slice; We try to go to 2D layers as fast as possible.
Since we compare the area we need to support based on subtracting layers from each other, we might incorrectly select certain areas if the resolution is different.

diegopradogesto, 2018-10-02:
In this specific case, you can try reducing the overhang angle. Other option you can use is to create a custom support (for instance using the _Custom Supports_ package in the Toolbox) in that part with the exact size you desire.

Ghostkeeper, 2018-11-16:
I think this can be solved (with some work). We compute the overhang distance by taking the tangent of the overhang angle and multiplying that with the layer height. We currently always use the layer height setting, but there it should use the actual height of the layer.

This sort of problem is everywhere in the engine. The adaptive layer height implementation is vastly incomplete in that regard, which is why it's in Experimental.