Ultimaker/Cura #4398: [3.4.1]

Liger0, 2018-09-21:
**Application Version**
3.4.1

**Platform**
Windows 10 x64

**Printer**
Creality cr-10s

**Steps to Reproduce**
Increase the infill speed from let's say 40 to 70

**Actual Results**
The time needed for the print results to be higher of about 15 minutes.

**Expected results**
The time needed for the print should result lower.

[CCR10_Blade_Middle_X3.zip](https://github.com/Ultimaker/Cura/files/2405609/CCR10_Blade_Middle_X3.zip)

rburema, 2018-09-21:
Hi @Liger0,

I'm using the same setup (Windows 10/x64, used your model, selected your printer(s)), and I can't reproduce this.

I tried both the _Creality CR-10 S4_ and _Creality CR-10 S5_, but it's always 7h42 min for 60 mm/s infill speed and 8h21m for 40 mm/s infill speed.

Would you kindly recheck if you put all the other options on the same values?

cheers, Remco.

Liger0, 2018-09-21:
> Hi @Liger0,
>
> I'm using the same setup (Windows 10/x64, used your model, selected your printer(s)), and I can't reproduce this.
>
> I tried both the _Creality CR-10 S4_ and _Creality CR-10 S5_, but it's [...]

Hi there @rburema.
I will attach the screenshots, as you see with more than double infill speed it will need about 15 minutes more. No other changes are done over the infill speed.
[image]
[image]

BagelOrb, 2018-10-04:
I think there was a fix to the time estimates recently, right @ghostkeeper?

nallath, 2018-10-08:
Can't reproduce this on master, so I'm going to assume this is fixed.

closed 2018-10-08 by nallath