Ultimaker/Cura #4401: [3.4.1] Cura making wrong GCODE with "Scalable Extra Prime"

digiTALker, 2018-09-21:
**3.4.1**

**Win7x64**

**RepRap Repetier  firmware**

[CFDMP_EGG_SEPERATOR_V3.zip](https://github.com/Ultimaker/Cura/files/2406832/CFDMP_EGG_SEPERATOR_V3.zip)

**Steps to Reproduce**
Check if Scalable Extra Prime are enabled.
[image]
Put script to end of GCODE
[image]

**Actual Results**
Open generated GCODE and look at the part with END-script.
[image]
And instead of retraction we get extruding.

**Expected results**
Correct END SCRIPT

**Additional Information**

fieldOfView, 2018-09-21:
Scalable Extra Prime is a plugin by @Pheneeny, the best is to open an issue over at its own issue queue:
https://github.com/Pheneeny/CuraPlugin-ScalableExtraPrime

A hint to the author of the plugin (which you may want to add to the issue) is that the plugin should not process the last "layer" in the gcode_data.

Pheneeny, 2018-09-22:
This should be covered by this issue: [Retraction in End G-Code also modified by plugin](https://github.com/Pheneeny/CuraPlugin-ScalableExtraPrime/issues/6)

I unfortunately didn't see that issue until today, but I'm currently working on a fix.

This issue can probably be closed on this repository.

closed 2018-09-27 by nallath