Ultimaker/Cura #4403: [3.4.1] missing line on internal wall with stock profiles

Liger0, 2018-09-21:
**Application Version**
3.4.1

**Platform**
Windows 10 x64

**Printer**
Cr-10S

**Steps to Reproduce**
Use any of the stock profiles

**Actual Results**
You will se at the center there will be a missing line so the internal is visible, I think it is caused by the line being smaller than the line width

**Expected results**
I think it should add an additional line at the center to fill the hole. The same goes on sharp edges that are just empty areas. I think there should be an optional over extrusion for them, since it is better to sand  the residues than having a missing edge on the print, or if there is I can't find it.

[image]
[image]
[image]

smartavionics, 2018-09-22:
With regard to that last image, the small narrowing of the line in the pointy region is not going to make any visible difference as the latency in the extruder/hotend will complete average out the small reduction in extrusion rate.

Liger0, 2018-09-22:
> 
> 
> With regard to that last image, the small narrowing of the line in the pointy region is not going to make any visible difference as the latency in the extruder/hotend will complete average out the small reduction in extrusion rate.

It actually makes an HUGE visible difference as I printer multiple items with that shape, and it is a real horrible effect.

smartavionics, 2018-09-22:
> It actually makes an HUGE visible difference as I printed multiple items with that shape, and it is a real horrible effect.

Really? I find that hard to believe. Anyway, just turn off the wall overlap compensation and it won't reduce the extrusion rate there.

Liger0, 2018-09-22:
> 
> 
> > It actually makes an HUGE visible difference as I printed multiple items with that shape, and it is a real horrible effect.
> 
> Really? I find that hard to believe. Anyway, just turn off the wall overlap compensation and it won't reduce the extrusion rate there.

It makes the conjunction point far better for real, that's why I will ask if you will be able to put it disable on the stock profiles with the next releases of Cura, I already read online various people complaining about related issues before finding out it was the cause of it, so making it available but not automatically checked would help the new entries.
Also, an option to fill the vertices would be helpful, since sanding it to have a sharp edge is better than having an underextrusion effect on it which you can't do anything about.

Liger0, 2018-09-22:
> 
> 
> I still don't believe this is the cause of your issue.
> Anyway, you will have to discuss your request to have the setting disabled by default with the Cura developers so just keep the issue open and someone will respond.

The conjunction point on the print is visible exactly how it is in the model, and is solved disabling that option, so yeah it is what causes it.
I hope there will be added a function to make the sharp edges printable in all the directions, both adjusting the line height automatically or something along this, or overextruding so it is possible to sand it.
I also have underextrusion everywhere along the infill shape on the external wall, or in case there are holes inside the model like the 2 holes in the model here. Even with 5 perimeters and 1% of overlap. I see other people online mentioned this exactly issue started after the 2.6.1 version of Cura, however I didn't try it but never found a solution tweaking the settings. I am going to try with other slicers soon and see if it is the same or not to understand if it is an hardware or software fault, since I changed my firmware and changed belt, screws etc positions without a minimal difference.

Ghostkeeper, 2018-11-21:
If you don't want the sharp edge to become thinner there, disable the "Compensate Wall Overlaps" setting. It's made thinner to prevent underextrusion due to the first line overlapping with the second line that goes back in almost the same direction.

I would expect the small gap filling technique to fill out the gap in your skin there, as it is too thin for an additional wall. But apparently it's not happening (as I can see in your screenshot that the feature is enabled).
Do you still have that model that reproduced the issue?

Liger0, 2018-11-21:
> If you don't want the sharp edge to become thinner there, disable the "Compensate Wall Overlaps" setting. It's made thinner to prevent underextrusion due to the first line overlapping with the [...]

As I see it seems the gap is so tiny that the fill gap option is not even engaged. It just happens on all the models, I would like to think this feature to be improved.
Here there is a typical result:
[image]

And here the model of the original part.
[wall gap.zip](https://github.com/Ultimaker/Cura/files/2604625/wall.gap.zip)

closed 2020-06-20 by no-response[bot]