Ultimaker/Cura #4404: Package installation scheduled, but not done on restart.

thopiekar, 2018-09-21:
Hey guys,

I received an alpha build about one week before from @Appesteijn. Its timestamp is: 20180907040418
After updating my packaging script, so I can make nightly builds from my CI easier - https://github.com/thopiekar/CuraPluginOven/tree/master-creating-packages .
When dragging a curapackage file into Cura, I get notified that it will be installed upon next start. So far so good, but it does not seem to work. Additionally, nothing in the logs mentions that is has failed, been done or whatever.

[cura.log](https://github.com/Ultimaker/Cura/files/2407241/cura.log)

* https://thopiekar.eu:5443/download/_dev/CuraSolidWorksPlugin-0.5.5.curapackage
* https://thopiekar.eu:5443/download/_dev/CuraSolidWorksPlugin-0.5.5.umplugin

Any ideas? Known bug? Already fixed? :see_no_evil:

Appesteijn, 2018-09-22:
Usually this means it crashes when it is loaded or it is missing dependencies. But I don't see anything in the log

thopiekar, 2018-09-22:
Ok, I can try it. I don't remember wrong this is what @nallath once mentioned via email.
If this really fixes the problem, then the package manager should better look into the package and it's package.json instead of looking at its name to determine the package_id of the plugin inside.

thopiekar, 2018-09-22:
Hmm, the CLI should give output then. But for some reason, it is really quiet.
Sadly it does not give me a hint about an option, which takes it into a verbose mode, but will look into the code later.
[image]

So back to @Appesteijn 's suggestion. What I noticed is that the package is missing in the cache. So I copied it manually into the cache and removed the version from the package's name, but no actual difference.
The log doesn't show any crash. Therefore I clicked on "Uninstall", but the package file is kept in the cache. That makes sense because it is stored within a cache, but a click on "Uninstall" is an explicit order by the user to remove it.
So either there is a bug or a feature :man_shrugging: 

So began to install my package by drag-and-drop again, but with the correct file name, so no version included. Again, the same behaviour as with the version included. Nothing in the cache and the plugin is disabled. No real error messages everywhere.

So the last option is that my package is broken and/or not following the conventions, right?
Looked into the toolbox's code and downloaded the CuraDrive package manually and drag-and-dropped it manually into Cura, as I did with mine.
.. Guess what: Same behaviour with a plugin, which should definitely work.
Looks like the manual installation via drag-and-drop is generally broken.

Besides that: Is it possible to place the plugin into the plugin directory of the config directory as we did in the past?

Appesteijn, 2018-09-22:
Drag and drop certainly worked on yesterdays 3.5 build, as I tested a lot of packages that were send for review. I can't access these builds now.
I think something goes wrong when Cura tries to install your plugin because I also see no folder being created in the plugins directory. You should still be able to manually copy the plugin into the plugin directory (that is what we also do when drag&drop installing)

thopiekar, 2018-09-22:
Ah, good to know. Well, I'm curious what your investigations will bring up on Monday or whenever you have time of course.

thopiekar, 2018-09-26:
Well, I think I see why my package fails. It misses this [Content_Type].xml and _rels stuff.
[image]

Oh, and noticed you increased the API again.
https://api.ultimaker.com/cura-packages/v1/cura/v6/packages

Well, I think I'll just take a less critical plugin, like the FreeCAD one, and update the package on your portal (what I would do anyways).
However, would like to send the packages around to the community first for intensive testing (eg. Autodesk Inventor) before distributing them in the toolbox.
That is why I put this effort in this package building script, you know.

Well, if you like to do some testing already, @Appesteijn , you can unzip the content from the package you can find under the links above.
Also I mirror my stuff on GitHub, but might be that things are out of sync. Additionally, just noticed that the files are not visible for public on my main repo on my NAS.
Will change that as soon as I have time of course. But want to fix that here before.
I'm really limited in time currently.

thopiekar, 2018-09-26:
Ok, tried to expand my script with the missing bits, but no chance.
I mean, I double checked and my package should behave like the recent OctoPrint package you sent me.

However, I copied the plugin manually into [config dir]/plugins, but there is a missing module. Looks like you shrinked Cura a bit, so it is gone. Will open another ticket about it.
Would be nice to get it for these lines: https://thopiekar.eu:5443/Cura-CAD-integration/CuraSolidWorksPlugin/commit/7b7cac3e25a8ac4b406531f8a5162d5f8698ddc8

Appesteijn, 2018-09-26:
Did you forgot to add the COM library? I added the 3rd_party directory from the v4 plugin, renamed the directory, zipped it and then I could upload it in the staging environment of the contribution portal.
The resulting curapackage can be drap&dropped in Cura 3.5. I can't test further as I have a Mac.

[CuraSolidWorksPlugin.zip](https://github.com/Ultimaker/Cura/files/2418826/CuraSolidWorksPlugin.zip)

thopiekar, 2018-09-26:
Ah, when downloading the code, I guess it is not caring about the submodules. To download everything completely for sure, you need to clone the repo and init the submodules.
So I should be also able to download a package from your portal, before publishing it into the wild?

Appesteijn, 2018-09-26:
Yes if you click the cloud icon you get the package.
[image]

thopiekar, 2018-09-26:
Can't get it working over here. Does't matter whether I upload everything in a zip incl or excl package.json. Also created a zip with CuraFreeCADPlugin directory in the root or not.

[image]

I can't help the portal as long as it does not tell me what is wrong :disappointed: 

This is the last zip I tried: https://thopiekar.eu/download/_dev/CuraFreeCadPlugin.zip

Appesteijn, 2018-09-27:
The Package ID and the zipped folder are case sensitive. If you rename the folder to CuraFreeCADPlugin and then zip it. It should work.

thopiekar, 2018-09-30:
Well, can create testing packages with your portal now.
Optionally, I will make my build script work, too, but that is my problem then, I guess. :wink: 

Therefore closing it for now.

closed 2018-09-30 by thopiekar