Ultimaker/Cura #4405: [3.4.1] Infill resumes after a few layers where not needed anymore

Liger0, 2018-09-22:
**Application Version**
3.4.1

**Platform**
Windows 10 x64

**Printer**
Cr-10S

**Steps to Reproduce**
Put an hole in the top part of the model

**Actual Results**
The infill stops at a certain height and resumes and stays for a few other layers.

**Additional Information**
Look at the layer number and you will see it correctly stops after it encounters the top walls and then randomly resumes.

[image]
[image]
[image]

smartavionics, 2018-09-22:
Hello, please attach the project file, without that, it's impossible to investigate your issue. Thanks.

Liger0, 2018-09-22:
> 
> 
> Hello, please attach the project file, without that, it's impossible to investigate your issue. Thanks.

[CCR10_Blade_Middle_X3.curaproject.zip](https://github.com/Ultimaker/Cura/files/2407796/CCR10_Blade_Middle_X3.curaproject.zip)

Here it is.

smartavionics, 2018-09-22:
It's doing that because you have the "alternate extra wall" setting enabled.

Liger0, 2018-09-22:
> 
> 
> It's doing that because you have the "alternate extra wall" setting enabled.

Oh.. got it, I never really understood that setting to be honest.
Before closing this ticket I will take the chance to ask you what causes the divided layers in the infill too, which is an effect similar to the previous one, just it makes the lyers divided an is not related to that option. It is a very creepy isssue or freature since with it the infill is extremely weak.

[CCR10_Guard_Spike.curaproject.zip](https://github.com/Ultimaker/Cura/files/2407907/CCR10_Guard_Spike.curaproject.zip)
[image]

BagelOrb, 2018-10-04:
I don't understand your last issue.

Perhaps you simply need to switch you infill pattern from lines to grid...

Ghostkeeper, 2018-11-29:
The second issue is caused by the Infill Layer Thickness setting. The Infill Layer Thickness is set to about twice the layer height, so it will only print infill every other layer (with double the flow).

closed 2018-11-29 by Ghostkeeper