Ultimaker/Cura #4406: Cura 3.4.1 crashes upon using arrow up on keyboard

icerabbitje, 2018-09-22:
**Application Version**

Cura 3.4.1
**Platform**

OS X 10.12.4
**Printer**

Creality Ender 3
**Steps to Reproduce**

New project, open file, layer view, press arrow up
**Actual Results**

Cura Crash
**Expected results**

Expect model to be built layer by layer. 
**Additional Information**

Related issue is that in layer view there is no height scroll bar shown, as there should be, which prompted me to press arrow up, figuring the layer model could be built/viewed that way.

alekseisasin, 2018-09-25:
Hi @icerabbitje , It is very strange. I do not have this issue. Might be while installation some packages were installed properly. To be sure I would as you to reinstall Cura or you can download Cura 3.5 Beta and test "arrow up". If you will have again the crash, then could you please share the log file.

icerabbitje, 2018-09-28:
I'm new to Cura. No add-on packages. 

Went from 
3.4.1. Open any Project. Switch from Solid view to Layer. (without preparing) Press arrow up. Crash. 

3.5.0 Beta from 9/18. Switch from Solid view to Layer or XRAY (without preparing) Press arrow up properly tilts the project plane. Other arrow keys help rotate etc. 

[ and the layer sliders show after using prepare, that was my novice error / lack of instruction to get that working ]

alekseisasin, 2018-10-01:
It is still hard to understand the reason of the crash without the log file. Please share the log file.
You can find it here:
The Cura GUI log file, located at
$USER/Library/Application Support/cura/<Cura version>/cura.log (OSX)

icerabbitje, 2018-10-01:
Found it. 
Had to enable the viewing of hidden files and folders.
[cura.log](https://github.com/Ultimaker/Cura/files/2435669/cura.log)

Interesting read, there's a lot of debug, warning, etc going on.

alekseisasin, 2018-10-03:
@LipuFei , can you take a look?

LipuFei, 2018-10-09:
At the end of the log file, it says:
```
An uncaught error has occurred!
Traceback (most recent call last):
   File "/Users/ultimaker/build/3.4/build/inst/lib/python3.5/site-packages/UM/Qt/Bindings/PointingRectangle.py", line 105, in updatePaintNode
OverflowError: argument 2 overflowed: value must be in the range -2147483648 to 2147483647
```

I guess that's the one.

LipuFei, 2018-10-09:
From the description, my guess is that something went wrong in https://github.com/Ultimaker/Cura/blob/a7f421c60cfc600ba1ee18a274d9c6ea1a868e1b/plugins/SimulationView/SimulationSliderLabel.qml

Ghostkeeper, 2018-11-23:
We have been unable to reproduce this so far. We can only say that the PointingRectangle got an invalid Y position from the log, but how that came to be is a mystery.

Ghostkeeper, 2018-12-13:
Let's hope that it won't show up again with the user either, then.

closed 2018-12-13 by Ghostkeeper