Ultimaker/Cura #4411: [3.4.1] Changing name doesn't apply if you click Save directly after

samyk, 2018-09-22:
**Application Version**
3.4.1

**Platform**
macOS 10.13.6

**Printer**
Ultimaker 2+

**Steps to Reproduce**
Click the pencil icon next to the name of the file at bottom right of Prepare window. Type a new name. Click "Save to File" or "Save to Removable Drive".

**Actual Results**
It will attempt to save with the **old** name.

**Expected results**
It should save with the new name typed in.

**Additional Information**
This an be avoided by clicking out of the edit area once you've renamed it *before* clicking Save. Once you've left the area, the change is applied, then hitting Save will use the new name.

alekseisasin, 2018-09-25:
Hi @samyk , I could not reproduce this issue. Do you do maybe some other steps before saving?

fieldOfView, 2018-09-25:
That's because it has been fixed in 3.5

alekseisasin, 2018-09-25:
I know there were changes related to changing project name. I checked both 3.4 and 3.5 and could not reproduce, for this reason, I started to doubt whether there some extra steps to reproduce or not.
Anyway, I assume the issue is fixed in Cura 3.5 release.

closed 2018-09-27 by alekseisasin