Ultimaker/Cura #4420: [3.4.1] Cura tree supports start "in the air"

biz2000, 2018-09-24:
**Application Version**
3.4.1

**Platform**
Windows 7

**Printer**
CR10

**Steps to Reproduce**
Load the trip-ribs1.stl file of this project (also attached)
[Uploading treesupportbug.zip…]()

https://www.thingiverse.com/thing:2071610

**Actual Results**
Some support start in the air, not linked to the main base standing on the bed.

**Expected results**
Cannot print this way for obvious reasons

**Additional Information**
Screenshot attached.

rburema, 2018-09-26:
Hi @biz2000, Could you check if the 3.5-BETA works for you? If so, this might be a duplicate of another (fixed) issue. cheers, Remco.

diegopradogesto, 2018-10-02:
Probably duplicate of #3249. That is fixed in the 3.5 version.

closed 2018-10-02 by diegopradogesto