Ultimaker/Cura #4422: Inner walls on in thin air while bridging when supporting walls are thin

soswow, 2018-09-24:
**Application Version**
3.4.1

**Platform**
Mac

**Printer**
wanhao duplicator i3
[PI3_Tie Hex Unibody.3mf.zip](https://github.com/Ultimaker/Cura/files/2411032/PI3_Tie.Hex.Unibody.3mf.zip)

**Steps to Reproduce**
Have a hollow space that is closed with a cap using bridging with wall to be just 1 layer thin, but max number of walls is 3.

**Actual Results**
It tries to print walls in thin air

**Expected results**
First hanging in the air layer should not have walls but one that supports this layer.

**Additional Information**
Just before cap. See there is only one line as a wall
[image]

And here is a bridging layer.
[image]

smartavionics, 2018-09-24:
Yes, it does that. I don't immediately know how to fix this. Will think about possible solutions. Thanks for the report.

Ghostkeeper, 2018-11-21:
You can set the Wall Line Count setting to 1, right? That should fix it fine for this model. In the case of other models, you can add a modifier mesh to add some extra walls here and there.

If we'd be removing walls in places where bridging occurs, you're going to see some very nasty banding in those places. I don't think that's going to be a good solution.

I don't think there is a good solution for cases like this. At least, not one that will work automatically.

theWaldschrat, 2019-09-15:
I ran into the same problem when trying to use sacrificial walls to avoid support. Traced it down to bridging from walls not larger (equal isn't enough) than the configured "Wall Thickness".
In this file one model has a wall exactly equal to "Wall thickness", the other is 0.01 mm thicker:
[Bridge-Test.zip](https://github.com/Ultimaker/Cura/files/3613582/Bridge-Test.zip)

Even worse, even if the wall is thick enough to start bridging, if you look at how it prints the bridge, the additional perimeter the bridge lines attach to is printed in thin air.

Ticking "Enable Bridge Settings" causes it always do bridging, no matter how thin the wall is, but as long as the wall is thinner than "Wall Line Count" plus "Extra Skin Wall Count", it'll still put the perimeters on top of nothing.
Test model with extra thin wall: [Bridge-Test-044.zip](https://github.com/Ultimaker/Cura/files/3613611/Bridge-Test-044.zip)

All in all I think it would be a good idea to start by not applying "Extra Skin Wall Count" to bridges. I think it will very rarely help but it does break things for some models.
To make sacrificial walls possible a setting like "Minimal Bridge Wall Count" would be helpful. Maybe hide it in "Enable Bridge Settings" and set it by default to equal "Wall Line Count". If the user wants, she can set it to 0 or 1 and sacrificial walls will work.

allyourcode, 2019-10-17:
This bug seems to affect 4.3.0 (the current version?).

smartavionics, 2019-10-17:
> This bug seems to affect 4.3.0 (the current version?).

Yes, because nothing has changed WRT this issue.

allyourcode, 2019-10-17:
> > This bug seems to affect 4.3.0 (the current version?).
> 
> Yes, because nothing has changed WRT this issue.

Are any changes on the way?

smartavionics, 2019-10-17:
Well, it's not a high priority for me, I can't speak for the UM people.

Ghostkeeper, 2019-10-18:
We don't really have an idea yet to fix this issue good. It is currently all expected behaviour but not necessarily desired behaviour. To make this work for all types of models is pretty hard.

attacktwinkie, 2020-04-30:
Having bridging issues exactly like this one, using Cura 4.6
Hoping there was a fix for this,   I've experienced 2 extra inner walls which results in massively failed bridgings.   Is there a way to start the bridge over the wall instead of over a floating inner wall?

attacktwinkie, 2020-04-30:
I did find that for models that only have 2 walls, but the slicer has a 4 wall setting would produce floating inner walls to reach that 4th wall.  setting walls to 2 resulted in the bridging starting on the true inner wall not the inflated inner 4th inner wall.

kind3r, 2020-05-07:
So, yeah, it's related to the number of walls which depending on your model can add air printed walls. I'm usually testing my bridge settings on this simple one layer test https://www.thingiverse.com/thing:2614805. But since my default is to have 3 walls Cura decides to do another inner air wall when reaching the bridging layer resulting in ... well [this](https://lh3.googleusercontent.com/dDqnUpSo0SoBiTQ-jS-Gy-aLpcYcF1NqbcNIHDNAAZNLqZ3aHcoqYkZcu5t1v_TXjX16rPKPTXwdiy4JAaXlChdZs8MvI12ZlgU8VY5FPeC8E14C2Ta1cqKudvRD05IwptE0jQ_WKPz_iQ91z7_W8c2OxHzDLX7HfEQKZXZ_0ATsteIIQD05Ub3ElZfbTUjDYCG-x2vdsoy8zWtHSbnhxTQ07GQQ2zy_xfFzj0Rn0sCnmMeRKIWi40dnHApyh3Y6MOyMCxE_Eo_p8KnHRNhThkJRUvHfL35UtnkyGQw-hfrZP1c0Lcpkc8q-F0pGwIrV2PQ1UuO2qs-JU0hPpR2Bs7VXAZ5TxPY7okfMxb0FsoeBmRGmSatEdFC6kal_7BLpDjyAk0SAu5jKJUIrzOD8JkMCjSw2zlPJ3hgxAWTk8JGoPvE8aN1kaaHPzCE0WbKZdzsPjmMa46ly7I1BIkR0dcnw3aURUITVkKRfs8qbkh3NPZyUcZP0iVg74ifYf4mzdWBNT4KllSIplRPlnbuW7IO3GxCnMQNFfwp2o4LShjhbkuERsLceWdukifHz_yNYZDfeHnbz06PIJFnbNdlpWIT5iKhNqWnvkK2LGJ_T9mNL5ZRZkbzhRzTo8PSknVMO4kMAkfv7u-Pbv7_57axFKDgRczu-hpLw6O_rQ8a2JV6y6prL3NEd-pMjZWDk-d0=w2128-h1596-no) (left side 3 walls, right side 2 walls).