Ultimaker/Cura #4424: [3.3.1] Can't save and load support blocks -- any workarounds?

fusentasticus, 2018-09-24:
Positioning a support block may be challenging. So I'm trying to save this work -- for when I later print a new version of my model. 

But `Save as` STL, followed by later `open` loads the block as as something to be printed, not as a support block.

Is this a bug? Or, maybe I'm asking for a feature request? Searched around -- didn't find answers.

Thx

fieldOfView, 2018-09-24:
STLs can only have single models, and cannot distinguish between a normal model and a support mesh.
Save your files as a "project", and they should retain all models and settings.

PS: do you realize that you can use the move, rotate and scale tools on the placed support blocks?

ChrisTerBeke, 2018-09-24:
@fieldOfView is right here. To save all settings associated with Cura, save it as project. Otherwise you loose Cura specific data. In the new version, we've made this more clear by calling other options 'export' instead of 'save'.

closed 2018-09-24 by ChrisTerBeke