Ultimaker/Cura #4425: [3.4.1] Lines done multiple times on each layer

Liger0, 2018-09-24:
**Application Version**
3.4.1

**Platform**
Windows 10 x64

**Printer**
Cr-10(S)

**Steps to Reproduce**
Use any model.

**Actual Results**
 Some lines are done multiple times. With circular edges even in spiral mode they are done in a very weird way, even leaving a gap.

**Expected results**
Only 1 pass for every line and every layer, not going back

**Additional Information**
(https://www.youtube.com/watch?v=d9alWZVv7dI)
[image]
[image]

alekseisasin, 2018-09-25:
Hi @Liger0 , So far I did not have such issue. In your video, I can see strange movements.  Can you try the same model with Ultimker 3? Also, you can share your project, then can we can validate Cura configuration settings.

Liger0, 2018-09-25:
> 
> 
> Hi @Liger0 , So far I did not have such issue. In your video, I can see strange movements. Can you try the same model with Ultimker 3? Also, you can share your project, then can we can validate Cura configuration settings.

I shared the project in the post already, I moved it under additional info, maybe now it is more visible.
Here there are 2 things that cause those issue.
-1. The preview bar is bugged.  Going to right with the cursor sometimes it goes back (like when you move to left the cursor), it is what causes troubles during the brim preview in the video (or skirt and raft).
-2. The wall itself is made in a really weird way. It happens with both custom and stock cr-10 profiles. It happens when there isn't any top or bottom layer as I see. With ultimaker 3 printer set it is the same.
Watch out what happens with 2 models and ultimaker 3 stock settings.
[image]
[UM3_testhshape.curaproject.zip](https://github.com/Ultimaker/Cura/files/2415487/UM3_testhshape.curaproject.zip)

Liger0, 2018-10-05:
Even if it's solid it still repeats each layer twice.