Ultimaker/Cura #4426: Not responding after some time [3.0.4-3.5]

Austin4705, 2018-09-25:
**Application Version**
3.0.4-3.5

**Platform**
Windows 10

**Printer**
Tevo Tornado 

**Steps to Reproduce**
Install Cura and have it not respond 

**Actual Results**
Cura doesn't respond and crashes

**Expected results**
CURA should open and work properly

**Additional Information**
When I install cura it works for a bit, then about 4-5 hours later it doesn't respond and crashes. Then closing the program or rebooting the pc doesn't work. After that, I uninstall the software  (which includes deleting all AppData for CURA as well) and install a new version. The same issue happens again.

alekseisasin, 2018-09-25:
Do you have the issue for all Cura version 3.0.4 to 3.5Beta? If yes, then I suspect the issue is related to your computer. Please check the virtual memory, maybe you do not have enough free space on hard drive, or some other application uses all computer resources, like your antivirus.

Austin4705, 2018-09-26:
I do, I have 20 GB free on my SSD and about 1tb on my hard drive. I don't use an antivirus, while I should. Do you think it could because I had recently upgraded my pc from intel to amd? If you have any ideas please tell me.

alekseisasin, 2018-09-26:
I cannot explain why it works only for a couple of hours and after restart no more. Only after installation. We do not have a similar issue.
In your log file, I am missing Initialized OpenGL subsystems. Do you have proper drivers for your graphical card?

LipuFei, 2018-10-09:
Hi @Austin4705 , did you do anything before you left it for 4-5 hours? Any specific thing/steps can help, thanks!

Ghostkeeper, 2018-12-13:
We could not reproduce this problem. We can't do anything to help fixing it unless we get more information.

closed 2018-12-13 by Ghostkeeper