Ultimaker/Cura #4427: [3.4.1] Hang after Splashscreen with White Window

rallant92, 2018-09-25:
**Application Version**
3.4.1

**Platform**
Windows 10 64bit

**Printer**
None

**Steps to Reproduce**
Start Cura

**Actual Results**
The splashscreen appears, reaches "loading interface", and then disappears.
Then the main window appears, maxed to the screen.  The title bar is fine (showing Ultimaker Cura) but the rest of the window remains white.  After a while "(Not Responding)" also appears in the title bar.

**Expected results**
The main window should draw and show an empty work area.

**Additional Information**
I have used this installation for several weeks fine but I stopped using it for a while.  Coming back it will no longer start successfully, ever.  I believe the relevant change is the graphics driver has been updated.  The CPU is an i7-8700K and I am using it's Intel(R) UHD Graphics 630 part.  The driver is the latest on the Intel website:  24.20.100.6286.  

The window is appearing maxed on the second monitor at 2560x1080.  It has done this many many times before without problem so it's less important.

The %AppData%\Roaming\cura\stderr.log ends with:
2018-09-24 23:32:08,724 - DEBUG - [MainThread] UM.Qt.QtRenderer._initialize [183]: Support for Vertex Array Objects: True
2018-09-24 23:32:08,728 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [92]: Initialized OpenGL subsystems.
2018-09-24 23:32:08,729 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [93]: OpenGL Version:  4.1.0 - Build 24.20.100.6286
2018-09-24 23:32:08,730 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [94]: OpenGL Vendor:   Intel
2018-09-24 23:32:08,731 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [95]: OpenGL Renderer: Intel(R) UHD Graphics 630
2018-09-24 23:32:08,733 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\default.shader]...
2018-09-24 23:32:08,739 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\selection.shader]...
2018-09-24 23:32:08,745 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\default.shader]...
2018-09-24 23:32:08,751 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\composite.shader]...
2018-09-24 23:32:08,762 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\overhang.shader]...
2018-09-24 23:32:08,772 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\striped.shader]...
2018-09-24 23:32:08,783 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\transparent_object.shader]...
2018-09-24 23:32:08,791 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\striped.shader]...
2018-09-24 23:32:08,804 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\default.shader]...
2018-09-24 23:32:08,811 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\grid.shader]...
2018-09-24 23:32:08,819 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\transparent_object.shader]...
2018-09-24 23:32:08,830 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\platform.shader]...

waiting a minute or two never adds anything else to the log.

rallant92, 2018-09-25:
I tried disconnecting the secondary display and success!  It appears maxxed on the primary screen now.

The splashscreen has always seemed to appear on the primary display.

I have often noticed issues when moving the window to the secondary screen that the menu items still appear on the primary screen.

Anyways, here the new end of stderr.log:
2018-09-25 12:38:47,237 - DEBUG - [MainThread] UM.Qt.QtRenderer._initialize [183]: Support for Vertex Array Objects: True
2018-09-25 12:38:47,241 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [92]: Initialized OpenGL subsystems.
2018-09-25 12:38:47,242 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [93]: OpenGL Version:  4.1.0 - Build 24.20.100.6286
2018-09-25 12:38:47,243 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [94]: OpenGL Vendor:   Intel
2018-09-25 12:38:47,244 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [95]: OpenGL Renderer: Intel(R) UHD Graphics 630
2018-09-25 12:38:47,245 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\default.shader]...
2018-09-25 12:38:47,252 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\selection.shader]...
2018-09-25 12:38:47,257 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\default.shader]...
2018-09-25 12:38:47,263 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\composite.shader]...
2018-09-25 12:38:47,274 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\overhang.shader]...
2018-09-25 12:38:47,284 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\striped.shader]...
2018-09-25 12:38:47,294 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\transparent_object.shader]...
2018-09-25 12:38:47,302 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\striped.shader]...
2018-09-25 12:38:47,314 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\default.shader]...
2018-09-25 12:38:47,320 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\grid.shader]...
2018-09-25 12:38:47,327 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.4\resources\shaders\platform.shader]...
2018-09-25 12:38:47,498 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-09-25 12:38:52,679 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 5.2s
2018-09-25 12:38:57,525 - DEBUG - [MainThread] cura.AutoSave._onTimeout [49]: Autosaving preferences, instances and profiles

Thank you for letting me get going again.  I never would have thought to disconnect the secondary display.

Playing around with it a bit, the steps to reproduce are:
1. Start Cura
2. Move the window to the secondary display
3. Close Cura
4. Restart Cura

Expected result:

See Cura window in the secondary display drawn in

Actual result:

See Cura window in the secondary display but it remains white and not responding.

fieldOfView, 2018-09-26:
Have you tried the 3.5 beta? It uses an updated GUI framework, and that might fix the issue.

Note that I use cura on my secondary display daily, with a very similar gpu, and it has never been an issue.

bheet, 2018-10-02:
I am running into the same issue on 3.4.1 and I am running on a Geforce 1080TI and 3 monitors. When Cura is starting you see the splash screen in my widescreen monitor running at 2560x1080. The interface tries to load on my 3rd monitor at 1920x1080 and it freezes on a white screen. Disconnecting the 3rd monitor does allow Cura to load. My guess is if Cura does detect the resolution then it is grabbing the wrong one on startup.

fieldOfView, 2018-10-02:
No, Cura tries to open at the same location and size that it was closed on. I don’t know why it has a problem restoring to a display it can succesfully be dragged to once started.

LipuFei, 2018-10-10:
Hi @rallant92 @bheet , wonder if you have tried 3.5? Does it happen to solve this problem?

bheet, 2018-10-11:
It did not fix it as I was able to recreate the failure with 3.5. When opening on my Main window it is fine. If I move the window to my alternate monitor and X out then it will try to open in the alternate window and not load. I have to unplug that monitor so Cura will load on my main and then I have to quit out through the menu instead of X out or it will keep trying to load on the other monitor.

[cura.log](https://github.com/Ultimaker/Cura/files/2467160/cura.log)

rallant92, 2018-10-11:
I generally avoid software that is not stable but I do want this fixed and the software is a late beta and Cura does allow multiple versions so I gave it ago.

The problem remains as last described.  Only unplugging the HDMI cable from the second monitor allows Cura to start successfully.  From that point on, never ever move Cura to the second monitor.  

There is a similar oddity with the window used for the menu items.

I am happy to run with diagnostic flags or a special binary to get more information.

bheet, 2018-10-12:
Using Quit instead of the X in 3.5 no longer corrects the issue. Always tries to open in the monitor it freezes up in the first time.

Ghostkeeper, 2018-12-19:
Try to move the window to the other screen before quitting (instead of X). It saves the window position upon quitting.

Ghostkeeper, 2020-06-10:
This is most likely an incarnation of this Qt bug: https://bugreports.qt.io/browse/QTBUG-50371

Updating to Qt 5.12 or newer would fix it if it is.

achugb, 2020-07-29:
https://github.com/Ultimaker/Cura/issues/7848#issuecomment-665721836

Ghostkeeper, 2021-05-13:
Going by https://github.com/Ultimaker/Cura/issues/9708, the Qt update didn't fix this yet.

usernametakenaaaa, 2022-05-28:
Should be re-enabled, it's the same bug again

fieldOfView, 2022-05-29:
@usernametakenaaaa have you tried unchecking the "Restore window position on start" option in the General preferences in Cura?