Ultimaker/Cura #4428: [3.5] Crash when deselecting model that has Print as Support mesh type.

fieldOfView, 2018-09-25:
**Application Version**
3.5 (beta)

**Platform**
Windows (likely any)

**Printer**
any

**Steps to Reproduce**
* Load an object
* Select object and Per Model Settings
* Change Mesh Type to "Print as Support"
* Deselect the object

**Actual Results**
Cura crashes when deselecting the object. Log entry: (edited for brevity)
```
An uncaught error has occurred!
Traceback (most recent call last):
  File "...\Cura\cura\CuraApplication.py", line 874, in event
    return super().event(event)
  File "...\Uranium\UM\Qt\QtApplication.py", line 411, in event
    event._function_event.call()
  File "...\Uranium\UM\Event.py", line 189, in call
    self._function(*self._args, **self._kwargs)
  File "...\Uranium\UM\Settings\Models\SettingDefinitionsModel.py", line 561, in _updateVisibleRows
    self.visibleCountChanged.emit()
RuntimeError: wrapped C/C++ object of type SettingDefinitionsModel has been deleted
```

**Expected results**
Not crash.

**Additional Information**
Other Mesh Types do not seem to trigger this crash, neither does 3.4.

alekseisasin, 2018-09-26:
It happens only on Windows.
DEV: CURA-5766

fieldOfView, 2018-09-28:
The Runtime Error is easily caught, and skipping the emit should be harmless if the object is no longer around:
```
@@ -570,7 +570,10 @@ class SettingDefinitionsModel(QAbstractListModel):
             del self._row_index_list[row]
             self.endRemoveRows()

-        self.visibleCountChanged.emit()
+        try:
+            self.visibleCountChanged.emit()
+        except RuntimeError:
+            Logger.log("w", "RuntimeError occured: wrapped C/C++ object of type SettingDefinitionsModel has been deleted")

     # Helper function to determine if a setting(row) should be visible or not.
     def _isDefinitionVisible(self, definition, **kwargs):
```

I just don't know why this started happening now.

Shall I make a PR out of the above fix?

alekseisasin, 2018-09-28:
We also do not know, probably after Qt version update. we already have a fix for this ticket, it is in QA:
https://github.com/Ultimaker/Cura/commit/2a201cef23b8e2620cecf70f84b42568702bee91#diff-0
https://github.com/Ultimaker/Uranium/commit/e9eb03602efc9e17565fc80abceaca610e1f430d#diff-0

fieldOfView, 2018-09-28:
I just noticed. I'll close this issue then.

closed 2018-09-28 by fieldOfView