Ultimaker/Cura #4430: [3.5.0 BETA] Missing distutils module in Ultimaker software distributions

thopiekar, 2018-09-26:
Hey guys,

in Cura 3.4 I was able to use the distutils module to run the following line:
https://thopiekar.eu:5443/Cura-CAD-integration/CuraSolidWorksPlugin/commit/7b7cac3e25a8ac4b406531f8a5162d5f8698ddc8

Any chance to get it back before the final release?

ChrisTerBeke, 2018-09-26:
@LipuFei was the distutils package removed from the build system?

thopiekar, 2018-09-26:
Well, read from the mails that we switched from python 3.4 to 3.5? Or whatever?

Haven't noticed that and that one is important to me, since I ship PyWin32, which is binary linked against libpython*.dll, and shipped with the packages. Asked to add PyWin32 to Cura by default a long time ago, but that didn't get any attention.
However, would be great if I could get distutils back or provide a function in the future, so I can add these switches without worries.

Additionally, would be great to ask Cura about the current (plugin) API level. Haven't seen any code so far, which does that.

LipuFei, 2018-09-26:
Not explicitly removed, but also not explicitly included either.

ChrisTerBeke, 2018-09-26:
Maybe distutils got dropped in the Python standard set from 3.5 to 3.6?

LipuFei, 2018-09-26:
Cura 3.5 is still using Python 3.5. I haven't managed to make 3.6 fully working on all OSes yet.

ChrisTerBeke, 2018-09-26:
@thopiekar I see you only use LooseVersion from distutils. It's quite easy to replace that with a comparison function that uses the UM.Version class.

LipuFei, 2018-09-26:
@thopiekar Unfortunately there's some problem with creating new build environments on the build machines lately, so I don't know if it can be added into 3.5 quickly.

thopiekar, 2018-09-26:
Ok, if there is a Cura/Uranium way to do it, I will try to go this way.

Thank you for the help and hints guys 😉✌️

thopiekar, 2018-09-30:
Forget about this ticket.
Looks like I was working on Linux with my PPA. So distutils was available through my Linux installation.
Checked the older Cura installations for Windows and distutils, so doubt is is ever there, even that I'm still sure that this code with distutils was working on Windows.

Anyway, let's close this.

closed 2018-09-30 by thopiekar