Ultimaker/Cura #4432: Material Compatibility

Lschmitendorf, 2018-09-26:
What's up the the material compatibility check in 3.4.1 for the Ultimaker 3 and 3 Extended?  This feature has essentially disabled the second extruder on my Ultimaker 3 Extended even though Ultimaker shows PLA and PVA as compatible materials because the Cura software deems the materials incompatible.

[image]
[image]

fieldOfView, 2018-09-26:
I think it wants you to use a BB core for PVA.

ChrisTerBeke, 2018-09-26:
> I think it wants you to use a BB core for PVA.

This

closed 2018-09-26 by ChrisTerBeke

Lschmitendorf, 2018-09-26:
I am using a BB cove for PVA, the printer will not let you load the material any other way.

ChrisTerBeke, 2018-09-26:
If you have your printer network connected Cura will sync the physical configuration of the printer.

Lschmitendorf, 2018-09-26:
The printer is network connected.  I manually change the second core in Cura and it is printing.  Letting it run and see what happens.

Thanks all for the quick feedback!