Ultimaker/Cura #4433: cura3.5--Unable to set material diameter.Unable to set material diameter.

robin1015, 2018-09-27:
Unable to set material diameter.

Unable to set material diameter.

nallath, 2018-09-28:
Please use (and fill in) the provided template; https://github.com/Ultimaker/Cura/blob/master/.github/ISSUE_TEMPLATE.md

Josephur, 2018-10-02:
I'm guessing your trying to modify one of the default materials instead of making a duplicate (copy) of it, and then setting the diameter on your copy.  This is the "proper" way to add a new material that is based on or shares properties with another material.  If your adding a material that is completely new then use the Create function instead.  That being said you really should following the template nallath posted if this is truly an issue for you and not "user error".

fieldOfView, 2018-10-02:
It used to be possible to set the material diameter in the sidebar. If you want a material diameter setting back in the sidebar, install the "Printer Settings" plugin from the Toolbox.

diegopradogesto, 2018-10-02:
I guess it's a custom printer definition with no extruder definition. We had this issue with material diameter and machine nozzle size. 

I'll close this ticket for now. @robin1015 you can reopen the ticket if you provide more information by correctly filling the template.

closed 2018-10-02 by diegopradogesto