Ultimaker/Cura #4434: Crash when changing WiFi

BagelOrb, 2018-09-27:
**Application Version**
master

**Platform**
Linux

**Printer**
UM3

**Steps to Reproduce**
- Be connected to an UM3
- change WiFi to a different WiFi a couple of times

**Actual Results**
Crash

**Expected results**
Any exception is caught and the zeroconf plugin is restarted (or whatever)

**Additional Information**
```
2018-09-27 10:25:22,181 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///home/t.kuipers/Development/CuraTK/resources/qml/Topbar.qml:22: TypeError: Cannot read property 'acceptsCommands' of undefined
2018-09-27 10:25:22,182 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///home/t.kuipers/Development/CuraTK/cura/../plugins/UM3NetworkPrinting/resources/qml/UM3InfoComponents.qml:109: TypeError: Cannot read property 'materialNames' of undefined
2018-09-27 10:25:22,182 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///home/t.kuipers/Development/CuraTK/cura/../plugins/UM3NetworkPrinting/resources/qml/UM3InfoComponents.qml:100: TypeError: Cannot read property 'hotendIds' of undefined
2018-09-27 10:25:22,182 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///home/t.kuipers/Development/CuraTK/cura/../plugins/UM3NetworkPrinting/resources/qml/UM3InfoComponents.qml:16: TypeError: Cannot read property 'acceptsCommands' of undefined
2018-09-27 10:25:22,182 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///home/t.kuipers/Development/CuraTK/cura/../plugins/UM3NetworkPrinting/resources/qml/UM3InfoComponents.qml:17: TypeError: Cannot read property 'authenticationState' of undefined
2018-09-27 10:25:22,182 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///home/t.kuipers/Development/CuraTK/cura/../resources/qml/SidebarHeader.qml:24: TypeError: Cannot read property 'connectedPrintersTypeCount' of undefined
2018-09-27 10:25:22,183 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///home/t.kuipers/Development/CuraTK/cura/../resources/qml/PrepareSidebar.qml:24: TypeError: Cannot read property 'acceptsCommands' of undefined
2018-09-27 10:25:22,185 - DEBUG - [MainThread] cura.PrinterOutput.NetworkedPrinterOutputDevice._createNetworkManager [271]: Creating network manager
2018-09-27 10:25:52,341 - DEBUG - [MainThread] cura.PrinterOutput.NetworkedPrinterOutputDevice._createNetworkManager [271]: Creating network manager
2018-09-27 10:26:20,070 - DEBUG - [MainThread] UM.Controller.setActiveStage [159]: Setting active stage to MonitorStage
2018-09-27 10:26:20,170 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///usr/local/lib/python3.6/dist-packages/PyQt5/Qt/qml/QtQuick/Controls/Styles/Base/ScrollViewStyle.qml:384: ReferenceError: __styleData is not defined
2018-09-27 10:26:24,340 - DEBUG - [MainThread] cura.PrinterOutput.NetworkedPrinterOutputDevice._createNetworkManager [271]: Creating network manager
2018-09-27 10:26:30,350 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///home/t.kuipers/Development/CuraTK/plugins/UM3NetworkPrinting/src/../resources/qml/ClusterControlItem.qml:523:33: QML Image: Failed to get image from provider: image://print_job_preview/3/6ec1b9bf-1c7b-4f46-b13d-e31283cfcfdf
2018-09-27 10:26:30,368 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-09-27 10:26:30,369 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-09-27 10:26:30,369 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/home/t.kuipers/Development/CuraTK/cura/../plugins/UM3NetworkPrinting/src/UM3OutputDevicePlugin.py", line 134, in _onDeviceConnectionStateChanged
2018-09-27 10:26:30,369 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self.getOutputDeviceManager().addOutputDevice(self._discovered_devices[key])
2018-09-27 10:26:30,370 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/home/t.kuipers/Development/Uranium/UM/OutputDevice/OutputDeviceManager.py", line 113, in addOutputDevice
2018-09-27 10:26:30,370 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self.outputDevicesChanged.emit()
2018-09-27 10:26:30,370 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/home/t.kuipers/Development/Uranium/UM/Signal.py", line 212, in emit
2018-09-27 10:26:30,370 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self.__performEmit(*args, **kwargs)
2018-09-27 10:26:30,370 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/home/t.kuipers/Development/Uranium/UM/Signal.py", line 315, in __performEmit
2018-09-27 10:26:30,370 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     func(dest, *args, **kwargs)
2018-09-27 10:26:30,370 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/home/t.kuipers/Development/Uranium/UM/Qt/Bindings/OutputDevicesModel.py", line 62, in _update
2018-09-27 10:26:30,371 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self.beginResetModel()
2018-09-27 10:26:30,371 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: RuntimeError: wrapped C/C++ object of type OutputDevicesModel has been deleted
```

LipuFei, 2018-10-03:
For dev, JIRA ticket CURA-5791

LipuFei, 2018-10-10:
Fixed in https://github.com/Ultimaker/Uranium/commit/989d98e56d3d2115bb4c9b5cfa4682670b6a6451

closed 2018-10-10 by LipuFei