Ultimaker/Cura #4435: [3.4.1] Lines of top and bottom shell are "paired"

MB53, 2018-09-27:
**Application Version**
Cura 3.4.1

**Platform**
Windows 7/ 64 bit

**Printer**
A8, Nozzle 0.4 mm

**Steps to Reproduce**
any model printed

**Actual Results**
The lines of the top and bottom shell (4 layer bottom and 4 layer top) are "paired" or "doubled" (except the lines of the wall). Every second line is to close (less than 0.4 mm) to the first line. And that is resulting in a gap between every second and every third line (distance much more than 0.4 mm). This is true for Y direction and X direction as well.
I attached 2 screen shots:

A) Cura Layer view 
- Brim 13 blue lines
- Wall 1 red and 2 green lines
- Bottom layer 29 yellow lines 
- Wall 2 green and 1 red lines

B) Microscope picture of printed PLA 
( from left to right)
- Brim 13 lines = OK
- 3 lines Wall = OK
- ONLY 15 visible and wide lines of bottom layer. But the printer is doing physically 29 lines/moves) = FAILURE
- 3 Wall = OK

**Expected results**
As Cura layer view, 29 lines and each is attached to the neighbor lines. No gaps.

**Additional Information**
Screenshot Cura layer view and microscope picture of printed pla.
[image]
[image]

alekseisasin, 2018-09-27:
Hi @MB53 , can you share the g-code of the test model?

MB53, 2018-09-27:
Hi,
 here it comes.

regards
Manfred

smartavionics, 2018-09-27:
Unfortunately, you can't supply files as email attachments, you have to come to this web page and attach them here.

MB53, 2018-09-28:
This is the g-code of the test model

[Riegel_Test.txt](https://github.com/Ultimaker/Cura/files/2427209/Riegel_Test.txt)

MB53, 2018-09-28:
done
regards

smartavionics, 2018-09-28:
Thanks for the gcode file.

Unfortunately, your skin lines are not running NS or EW so it's not so easy to check if the spacing is uniform by looking at the coordinates but here is a screen shot of another gcode viewer looking at your gcode. The skin lines do not look paired to me.

[image]

My guess is that it is a mechanical issue with the printer that is causing alternate lines to stick together.

BagelOrb, 2018-09-28:
This sounds very much like a mechanical issue. There is probably a little slack in your print head positioning system.

closed 2018-09-28 by BagelOrb

MB53, 2018-09-29:
Thanks for quick response.

yes the g-code looks OK. The problem seems to be the Printer. Because the brim and the walls are printed perfect, I would exclude the hardware. I'll try to change the printer firmware.

Thanks for your help
regards
Manfred

Ghostkeeper, 2018-11-23:
We've observed this also at work. It's not really a mechanical issue but more of a process or plastic issue. What happens is that the first line is much wider than the nozzle opening. That line then blocks part of the nozzle opening while printing the second line, providing counter pressure. The material then can't escape the nozzle opening fast enough, resulting in underextrusion. This underextrusion causes the second line to be thin enough that the problem no longer occurs for the third line, so the third line can print fine again with the appropriate thickness.

The fix for this is to make sure that the line width is not too thick (slightly larger than the nozzle size should be possible, but don't go too far) and the layer height is not too thick on the first layer.