Ultimaker/Cura #4438: [master] - odd skin lines appeared after merge of refactor_singleton_settingsbase

smartavionics, 2018-09-27:
**Application Version**

master branch

**Platform**

Linux

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

I have merged today's update to master (refactor_singleton_settingsbase) and now I am getting some odd skin lines appearing that have never been seen before.

[image]

[KMMB_bracket_flat_1.3mf-renamed.to.zip](https://github.com/Ultimaker/Cura/files/2425590/KMMB_bracket_flat_1.3mf-renamed.to.zip)

smartavionics, 2018-09-27:
Update - it's nothing to do with today's mega merge, it's the skin/infill optimization screwing up (yet again).

If I disable the call to optimizePaths() then the weird lines go away.

PS - I have to disable optimizePaths() when the gyroid infill is in use because it screws up with that also.

fieldOfView, 2018-09-28:
Omg, “that” pr merged?

smartavionics, 2018-09-28:
It's in master, I don't know if it is in 3.5. Seems to have a lot of problems.

BagelOrb, 2018-09-28:
Sounds like something which should be fixed asap @Appestein

rburema, 2018-09-28:
It might help if we have the definition for `kossel_mini_mb`, or otherwise the settings used to reproduce this on a (virtual) Kossel Mini.

smartavionics, 2018-09-28:
Doesn't that get included in the project file attached above?

Anyway, the printer definition is this:

````
{
    "id": "kossel_mini_mb",
    "version": 2,
    "name": "Kossel Mini (mb)",
    "inherits": "kossel_mini",
    "metadata": {
        "visible": true,
        "author": "Mark Burton",
        "manufacturer": "Other",
        "category": "Other",
        "file_formats": "text/x-gcode",
        "icon": "icon_ultimaker2",
        "platform": "kossel_platform.stl",
        "platform_offset": [0, -0.25, 0],
        "machine_extruder_trains":
        {
            "0": "kossel_mini_mb_extruder_0"
        }
    },

    "overrides": {
        "machine_gcode_flavor": {
            "default_value": "RepRap (RepRap)"
        },
        "machine_start_gcode": {
            "default_value": "G28 ; home all axes\nG1 X0 Y0 Z10 F8000 ; descend\n; run print prologue\nM98 P/sys/printstart.g\n"
        },
        "machine_end_gcode": {
            "default_value": "M83 ; extruder relative mode\nG1 E-5 F2000 ; retract\nM104 S0 ; turn off extruder\nM140 S0 ; turn off bed\nM106 S0 ; turn off fan\nG28 ; home\nM84 ; disable motors\n"
        },
        "machine_max_feedrate_z": {
            "default_value": 300
        }
    }
}
````

And the extruder definition is this:

````
{
    "id": "kossel_mini_mb_extruder_0",
    "version": 2,
    "name": "Extruder 1",
    "inherits": "fdmextruder",
    "metadata": {
        "machine": "kossel_mini_mb",
        "position": "0"
    },

    "overrides": {
        "extruder_nr": { "default_value": 0 },
        "machine_nozzle_size": { "default_value": 0.4 },
        "material_diameter": { "default_value": 1.75 },
        "retraction_amount": { "default_value": 4 },
        "retraction_speed": { "default_value": 30 }
    }
}
````

````

rburema, 2018-09-28:
Hi @smartavionics,
I've pasted those into the def.'s and your project file now loads correctly I believe. However, I can't reproduce this on current master? See attached screenie.
[image]

smartavionics, 2018-09-28:
OK, well it must be something that's unique to my Cura. Sorry for wasting your time.

rburema, 2018-09-28:
No problem. Better to report such things early (with a little risk of 'can't reproduce') than to live with it in an actual release.

smartavionics, 2018-09-28:
BTW, here's a couple of images that show what I get with and without optimizePaths() being called...

[image]

[image]

Furthermore, if I use zig-zag infill, then I don't see wacky lines even if optimizePaths() is called.

smartavionics, 2018-09-28:
[image]

smartavionics, 2018-09-28:
Sorry, I've just tried the vanilla master branch and it also produces the wacky lines so I'm reopening this. The problem is not specific to my version of Cura.

rburema, 2018-09-28:
@smartavionics Just to be absolutely sure: With vanilla master I assume you mean Cura (frontend), Uranium, CuraEngine and libArcus right? They've all been updated. Recent CuraEngine doesn't want to compile without the updated libArcus as far as I recall.

smartavionics, 2018-09-28:
Hmm, I am currently using a non-updated Arcus and my engine is compiling OK.

When I said vanilla master I was referring only to CuraEngine as I don't see how any changes in Cura or arcus could make a difference with this issue.

rburema, 2018-09-28:
Absolutely right, yes, CuraEngine is the important one here. Just wanted to make sure we where testing the same thing. But it's weird that you didn't need the updated libArcus. Have you tried a `make clean`?

alekseisasin, 2018-10-01:
If we are talking about this: gcode_layer.optimizePaths(gcode.getPositionXY());
Then it is related to _mergeInfillLines_, this method already introduced a bunch of issues and getting more and more complicated with many _if_ statements. This functionality was added not so long time.

@smartavionics , I think you are using not latest CuraEngine, after merging 'Singleton Settingsbase" the engine does not work with old libArcus.

smartavionics, 2018-10-01:
I can assure you that I am using the current master branch for CuraEngine:

[image]

alekseisasin, 2018-10-01:
The latest changes in it were related to this function:
bool MergeInfillLines::tryMerge(.....

BagelOrb, 2018-10-01:
Ok please revert that commit and put the issue for which the commit was
introduced back in TODO in JIRA.

BagelOrb, 2018-10-01:
Not ethat this should also be done on the 3.5 branch.

alekseisasin, 2018-10-01:
DEV: CURA-5777

@BagelOrb , can you upload your model?

BagelOrb, 2018-10-01:
No that won't be possible.

The model is just a cube, but the way the infill is generated is not
portable to master.

Vandrasc, 2018-10-11:
There is still an issue when gradual infill is used together with the cross/3d infill pattern, we will tackle that in a different ticket. 

[image]

BagelOrb, 2018-10-11:
Aren't those the lines which appear because of Connect Infill Polygons?

That's what I made a fix for in https://github.com/Ultimaker/CuraEngine/pull/843

Ghostkeeper, 2018-11-22:
That last bug is also fixed now, thanks to BagelOrb's fix.

There's different issues with the merging of infill lines still, but the ones here are fixed. So I'm closing this report.

closed 2018-11-22 by Ghostkeeper