Ultimaker/Cura #4439: Support speed should follow initial layer speed

dotps1, 2018-09-28:
**Application Version**
3.4.1

**Platform**
Windows 10 Pro x64 

**Printer**
Creality Ender-3 using Klipper

**Steps to Reproduce**
Setting a low initial layer, IE 25mm/s to get good bed adheasion, support speed does not follow when they are laid right on the build plate.  I want my support speed to be fast, like really fast, I don't care about they quality, but I want the first layer to obey the initial layer speed, so I get good, accurate adhesion to the build plate.

**Actual Results**
Initial support layer runs at what ever speed the support speed is at.

**Expected results**
The initial layer speed should over ride the support speed for the first layer.

**Additional Information**

BagelOrb, 2018-09-28:
Could you please attach a gcode file which shows the problem? (Rename the file to a .gcode.txt file and drag in here)

dotps1, 2018-09-28:
its not really an "Issue" per say, but here is a gcode file: 
[CCR10_Pickle.gcode.txt](https://github.com/Ultimaker/Cura/files/2428655/CCR10_Pickle.gcode.txt)

its more that the initial layer of support should follow the speed setting set for the initial layer
[image]
I want my support speed after the first lay to be fast, like it is, 100mm/s, but I want the first layer of support to go at the `initial layer speed`: 25mm/s so I get better sticking to the bed.

So, its either I put my support speed at 25mm/s for the whole print, or I end up with supports that fail all the time, because there isn't much surface area sometimes sticking to the bed.

BagelOrb, 2018-09-28:
I think the initial layer speed should also apply to support, so what you are describing is a bug. It's not supposed to behave in that way.

dotps1, 2018-09-28:
my thought as well, yeah, I know my first layer of supports are laying down way faster then 25mm/s.

BagelOrb, 2018-09-28:
Your whole first layer is in 25mm/s (1500mm/min) according to the gcode.
It must therefore be a firmware problem.

Please contact the company of your printer.

closed 2018-09-28 by BagelOrb

markouustalu, 2021-03-23:
Hi, i have the same problem with Cura 4.8 on OSX.
[image]

and the gcode generated for first layer is as follows:

;TYPE:SUPPORT
G1 F5400 E177.60026
G1 F3609 X270 Y264.65 E178.81842
G1 X273.75 Y264.65 E178.97432
G1 X273.75 Y260 E179.16765
G1 X226.249 Y260 E181.14251
G1 X226.249 Y255 E181.35038
G1 X273.75 Y255 E183.32525
G1 X273.75 Y250 E183.53312
G1 X226.249 Y250 E185.50798
G1 X226.249 Y245 E185.71586
G1 X273.75 Y245 E187.69072
G1 X273.75 Y240 E187.8986
G1 X226.249 Y240 E189.87346
G1 X226.25 Y235.35 E190.06678
G1 X230 Y235.35 E190.22269
G1 X230 Y264.65 E191.44084
G1 X235 Y264.65 E191.64872

F3609 is exactly 60.15mm/s

markouustalu, 2021-04-05:
never mind. fixed it. As you can see on my image, my number of slower layers is 0, changing it to 1 makes it work perfectly. Thank you myself.

Ghostkeeper, 2021-04-09:
@markouustalu That is actually a detail that we've overlooked. Settings that don't affect the print should be hidden. I've made a small adjustment to clear up the confusion for future users. Thanks for reporting that!