Ultimaker/Cura #4442: when QT_SCREEN_SCALE_FACTORS is set CURA UI is messed up

gbil, 2018-09-28:
Cura v3.4.1 
Arch linux

my QT_SCREEN_SCALE_FACTORS is set as follows for 1.4 scaling 

QT_SCREEN_SCALE_FACTORS='DisplayPort-0=1.4;DisplayPort-1=1.4;DisplayPort-2=1.4;HDMI-A-0=1.4;DVI-D-0=1.4;'

and this is how it looks
[image]

if I run cura as 

QT_SCREEN_SCALE_FACTORS="" cura

everything looks fine

alekseisasin, 2018-10-02:
Hmm, I changed my scale factor to 1.4 and Cura works fine. I am not sure can we do something with that.

fieldOfView, 2018-10-02:
@alekseisasin on what OS did you test this? Could by Arch Linux only.

alekseisasin, 2018-10-02:
I have tested on Ubuntu.

gbil, 2018-10-14:
Sorry for the delayed reply. Not sure what the underlining issue might be but Arch is very up to date while I've noticed Ubuntu is a few QT versions behind so it might be the behavior of QT_SCREEN_SCALE_FACTORS has changed in between. 
Current QT5 is 5.11.2 in Arch 
What else can I provide to help you troubleshoot this?

alekseisasin, 2018-10-14:
Cura uses QT 5.10.0 version

gbil, 2018-10-14:
Ok didn't check to be honest if it uses its own libs or not. In any case, any other way to provide more info for troubleshooting this?

alekseisasin, 2018-10-15:
Maybe try to use different Graphical User Interface, like GNOME.

gbil, 2018-10-15:
@alekseisasin as QT_SCREEN_SCALE_FACTORS is set by KDE how testing with another DE would help troubleshooting ?

alekseisasin, 2018-10-16:
I just guess. You can try.

Ghostkeeper, 2018-12-21:
I can reproduce this issue using Qt 5.11, but I think it's more dependent on the actual resolution of your screen than on the difference between 5.11 and 5.10. I have two screens, one having small pixels (1.25 pixel scale) and one having larger pixels (1.0 pixel scale). I then ran Cura using QT_SCREEN_SCALE_FACTORS set to 1.4 and got this on my small-pixel screen: 
[image]

But when I move the Cura window to my large-pixel screen I get this:
[image]

I think it's more of a bug in Qt than in Cura, by the look of the symptoms. You can see that the text in these buttons is drawn way lower than the background of the buttons, but in Cura's source code the text and background are anchored to the button's rectangular area regardless of how large that area is. It just looks like it renders text in the wrong location if we use native rendering.

Ghostkeeper, 2018-12-21:
Talking this over with a colleague, we don't think this is a common enough use case to spend time on to fix. Let's hope that a future release of Qt fixes it. I couldn't find a bug report for it yet though.

(I should report it, I know, but it's kind of a lot of work to make a small application that reproduces the problem...)

closed 2018-12-21 by Ghostkeeper

G-Pereira, 2019-08-23:
For me, Cura in arch linux (in contrast to windows) always has been a pain in the ass, the menus are too much distorted, only appimage works. So I guess this is a "not common" issue also happening to me.
[image]

CJ1169, 2019-08-24:
I can confirm this has also been plaguing me. Can't actually use the software in this state. Running Manjaro.