Ultimaker/Cura #4446: Combing mode not working when printing gear

Taiko1976, 2018-09-30:
**Application Version**
3.4.1

**Platform**
Windows 7

**Printer**
Selfmade RepRap

**Actual Results**
Nozzle moves outside printed area when printing gear, resulting in closed gear teeth from oozing

**Expected results**
When Combing Mode is set to "All" the description says that the nozzle will stay within the printed area.

[image]

[CFDMP_Part Studio 1 - gear.zip](https://github.com/Ultimaker/Cura/files/2431290/CFDMP_Part.Studio.1.-.gear.zip)

diegopradogesto, 2018-10-02:
It doesn't happen with a freshly new Custom FDM Printer with by-default settings (combing_mode = All). 
[image]

So I'm pretty sure there are others settings you defined that are creating this output. Please provide a Cura Project File (_File->Save Project_)

Taiko1976, 2018-10-03:
As requested:

[Untitled.zip](https://github.com/Ultimaker/Cura/files/2440744/Untitled.zip)

diegopradogesto, 2018-10-09:
The culprit is the value of the _Retraction Minimum Travel_. You set it to 10mm, indicating that no retraction is happening if the distance between points is less than this value.

However it should comb anyway, so it seems like a bug. For the moment, the workaround for you is to change this value.

Taiko1976, 2018-10-10:
Thank you, looking forward to the bug fix. I'll try the workaround, too, although I tried it before and opted against it because of grinding issued on the filament. But thank you for the hint.

Vandrasc, 2018-10-15:
Devs see CURA-5818

nallath, 2018-11-16:
This should be fixed for the next release.

closed 2018-11-16 by nallath