Ultimaker/Cura #4447: Long travel moves without retraction

Taiko1976, 2018-09-30:
**Application Version**
3.4.1

**Platform**
Windows 7 64-bit

**Printer**
Selfmade RepRap

**Actual Results**
When printing big parts the nozzle moves unnecessarily all over the place without retraction, thereby draining the nozzle by oozing. When continuing the print at the new position the nozzle is empty and it takes some time until the material is printed normally again.

**Expected results**
The nozzle should not be moved all over the place without need. When finishing one feature the slicer should continue with the next feature nearby not moving the whole length of the part to start with the next, even though the feature could also be started right next to the previous one (e.g. walls: After printing the inner wall with the gear rack it finishes on the left hand side of the part, then moves all the way to the right hand side to start on the outer wall (again, without retraction or anything, thereby draining the material from the nozzle). Instead the slicer should move the nozzle the shortest distance to the next feature, which would be to the left to start there with the outer wall.

**Additional Information**
[image]

[Part Studio 1 - gear rack B.zip](https://github.com/Ultimaker/Cura/files/2431297/Part.Studio.1.-.gear.rack.B.zip)

smartavionics, 2018-09-30:
That's what you get when using combing mode all. That's how it's designed to work.  The new 3.5 beta provides a setting that forces retraction for combed moves longer than a given distance. It also has a new implementation of the no skin combing mode that really does avoid combing over skin.

Taiko1976, 2018-09-30:
Please read my post again: "The nozzle should not be moved all over the place without need. When finishing one feature the slicer should continue with the next feature nearby not moving the whole length of the part to start with the next"
 ... the problem is not about "not retracting" but that the nozzle is moving all over the place unnecessarily to start on the next feature that could as well be started right beside where the last feature was finished. 

From my perspective there are two ways of handling this: Best would be to improve the slicing algorithm to prevent unnecessary long travels (which is the primary problem and would improve the overall situation)... or (as second option) to provide a feature that allows configuration of retraction behavior (which would really only be a workaround). 

Since there are already gazillions of features and - to be honest - just adding another feature for every problem will just leave your software unusable at some point because nobody understands your features anymore (it's hard enough already).

I'd advise to solve the problem itself instead of just adding another feature to work around the problem.

diegopradogesto, 2018-10-02:
Please provide a project file instead of STL+GCode, otherwise it's really difficult to reproduce your issue. Among this, please give more hints such as the layer in which you see the problem.

Loading your gcode, I think you mean this part in the first layer:
[image]

Going from A to B, it chooses to go all the way to the left, probably related with your settings. So, please provide a project file (in 3.4.1 you can save it in _File->Save project_).

Taiko1976, 2018-10-03:
As requested:
[Untitled.zip](https://github.com/Ultimaker/Cura/files/2440745/Untitled.zip)

Ghostkeeper, 2018-11-29:
> The nozzle should not be moved all over the place without need. When finishing one feature the slicer should continue with the next feature nearby not moving the whole length of the part to start with the next,

That's the problem: There is a need. Our algorithm works exactly how you're describing. When it's completed with one part, it looks for the next closest part to print next.

There are multiple reasons why it's moving to the other side that I see in your file:
* When starting skin, it wants to start from a place that prevents seams in your skin. It finished printing the wall in the back side, then chooses a location such that the skin can be printed with as few seams in the skin as possible, which happens to be on the other side. Due to combing, it moves all the way to the left and back to the right again.
* When skin on the first layer is almost done, it missed one corner on the other side. So it moves there, again avoiding collisions with the walls so then going to the left and back to the right. It then also proceeds with all the saw teeth that were missed.
* After completing the printing of the skin, it fills some gaps between the walls. For that it also needs to move to the other side again. It's important that this gap filling is not done halfway through printing skin, even if that produces fewer travel moves, because the gap filling produces a lot of interruptions in flow so your skin would not flow nicely otherwise.
* After filling the gaps in the walls, it moves to the Z-seam on the other side.

For these reasons I would say that this is expected behaviour. Though it might seem inefficient, it is really better for your print.

closed 2018-11-29 by Ghostkeeper